espruino/Espruino #2402: Bangle.js 2 ZigBee support

tinxx, 2023-08-14:
Are there plans to make use of nRF52840's ZigBee support?

The chip should be capable of ZigBee as well as Thread and Matter.[1]
Is it theoretically possible with Bangle.js 2 or are there restrictions, e.g., due to antenna design?

If possible, where would one start implementing it?

Thanks--

[1) nrf52840-soc-v3.0.pdf](https://nsscprodmedia.blob.core.windows.net/prod/software-and-other-downloads/product-briefs/nrf52840-soc-v3.0.pdf)

thyttan, 2023-08-14:
Mentioned here on the forum: https://forum.espruino.com/conversations/380585/

tinxx, 2023-08-15:
Hi, just to clarify: This ticket is about ZigBee, not Matter. The forum thread is about Matter, not ZigBee. :)

gfwilliams, 2023-08-15:
Hi - there aren't any plans to add it. As I understand it ZigBee uses the same 2.4Ghz frequency range as Bluetooth so the hardware should work fine as-is.

To get started, you'll likely need this SDK: https://www.nordicsemi.com/Products/Development-software/nrf5-sdk-for-thread-and-zigbee

There's a newer SDK based on Zephyr but if you intend to run the Bangle.js firmware on top that's probably not going to work as Espruino's bluetooth is build on Nordic's original, non-zephyr API.

Then, because you'll be messing around with firmware a lot you want to buy an nRF52840DK board and wire up the Bangle's USB charge cable as follows: http://www.espruino.com/Bangle.js2+Technical#swd

... and try out some examples - maybe first on the DK board and then see about getting them running with Espruino.

But I'll close this issue for now as it's not really a bug per se - if you've got questions it's best to ask on the forum as having a bunch of questions in the issue tracker makes it really hard to see what we should be working on

closed 2023-08-15 by gfwilliams

tinxx, 2023-08-15:
Fair, thanks for the info, much appreciated!

thyttan, 2023-08-15:
> Hi, just to clarify: This ticket is about ZigBee, not Matter. The forum thread is about Matter, not ZigBee. :)

Yes, but ZigBee is mentioned by @fanoush in this comment on the thread: https://forum.espruino.com/conversations/380585/#16725086

fanoush, 2023-08-15:
> Hi, just to clarify: This ticket is about ZigBee, not Matter. The forum thread is about Matter, not ZigBee. :)

The SDK provided by Nordic is for both ZigBee and Matter - "[nRF5 SDK for Thread and Zigbee v3.2.0](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsdk_tz_v3.2.0%2Findex.html&cp=8_7_2)" As that specific version  is based on SDK15 just as Espruino for Bangle2 there is some chance to get it working. Anyway, use that forum thread (or another) if you try it. Getting Zigbee working with that will be useful for Thread too.