espruino/Espruino #2409: ESP32 - digitalPulse() not changing pulse width

SimonGAndrews, 2023-09-05:
While attempting to use the servo module it seems most likely that the supporting Espruino function digitalPulse(0 is not working correctly on an ESP32 ( Firebeetle ESP32-E).

I have put a logic analyser on the pulsed pin and it seems the pulse width is always about 600us in response to a digitalPulse(D4,1,x) command no matter what the value of x. Where x should be the length of the pulse in milli seconds. 

Please see[ supporting discussion](https://forum.espruino.com/conversations/389271/) at https://forum.espruino.com/conversations/389271/

SimonGAndrews, 2023-09-05:
Happy to support with further testing ,  on different ESp32 boards and Espruino Pico if that helps.

gfwilliams, 2023-09-06:
Thanks - very strange. And longer pulses don't work?

This could potentially be related to `jstimer` changes that went in a while ago, where we moved to using relative times rather than absolute for timer tasks. I guess we could test before/after to see if that's what broke it

SimonGAndrews, 2023-09-06:
Lo

SimonGAndrews, 2023-09-06:
Thanks,  Ok I’ll try a version pre that change and longer pulses. I will also check my measurement method with a pico.

MaBecker, 2024-08-29:
@SimonGAndrews  Does this also occour with the latest firmware?

SimonGAndrews, 2024-08-29:
Hi , I will check and let you know .

SimonGAndrews, 2025-02-03:
Ive had another look at this and I believe digitalPulse() is working on the ESP32 Builds now.   Here is a test on the S3 

[image]

gfwilliams, 2025-02-03:
Excellent, thanks for checking! I made some big improvements for ESP32 a few months back so I'm glad this is sorted.

Someone was trying to control a light dimmer with it, and I think even with this it wasn't as accurate as he'd have liked, but I think we're getting as good as we can get with interrupts

closed 2025-02-03 by gfwilliams

SimonGAndrews, 2025-02-03:
And a similiar test on the ESP32.

[image]