espruino/Espruino #2417: Settings app not loading after update to firmware 2v19 (Bangle.js 2)

Moini, 2023-09-17:
Hi! 

I just found the fresh update for the Bangle.js 2 firmware in the AppLoader and installed it, as it seemed to promise improvements with heart rate detection (heart rate seems/seemed(?) mostly random - it's nearly the same when wearing the Bangle as it is when not wearing it, this becomes evident now that I can load the data into Gadgetbridge).

However, when I now choose the 'Settings' app in my menu, it does not stop displaying the 'Loading…' overlay (I waited for > a minute as this only took a second or two previously).

How can I help find out why Settings do not load? I would need access to those sometimes...

thyttan, 2023-09-17:
I didn't have this problem on 2v19 and don't on 2v19.15 either.

Do you maybe have minification turned on in the app loader?:
[image]

Did you try to reinstall the settings app?

gfwilliams, 2023-09-18:
This also seems to be mentioned on the forum: https://forum.espruino.com/conversations/390154/#comment17127881

But as mentioned there I can't seem to reproduce this at all.

gfwilliams, 2023-09-18:
I've just debugged this with Fabio's Bangle. In his case he'd turned on App Translations in the App Loader. Might you have done that too?

Moini, 2023-09-18:
@gfwilliams Yes, I have checked 'pretokenize', 'always update time' and 'German'.

It's not always active, as, for installing new apps, I'm switching between mobile phone (Gadgetbridge) where settings are kept (with those options checked), and Chromium, where settings are lost and I always forget to make them before I upload any apps (I think I've set cookies on auto-delete when closing or something like that).

@thyttan I have intentionally not done anything to try and fix it, so the cause can be found.

gfwilliams, 2023-09-18:
Thanks! Should be fixed now in the development app loader.

In general though, if you've ticked some option that says 'beta' next to it like the translations, it's *really* helpful if you can tell us, as there's a good chance it might be related.

closed 2023-09-18 by gfwilliams

Moini, 2023-09-18:
@gfwilliams Thank you! Aye, captain, I'll keep that in mind for my next reports :)

Is it possible for me to setup a notification somewhere, so I get notified when the live Website has been updated?

fparri, 2023-09-18:
Thanks again Gordon!

Moini, 2023-09-18:
Thanks, I'll check back in about a week from now anyway, to see whether there are any app updates, too :)

Moini, 2023-09-30:
Just found the hidden option to get the reinstall icon back, so was able to just reinstall the settings app without losing settings (didn't know whether uninstalling would delete the config along with it - not clear from the interface). 

Works again now!

gfwilliams, 2023-10-02:
> Just found the hidden option to get the reinstall icon back

Sorry, not sure I understand - what are you referring to here?

Moini, 2023-10-02:
I mean this one:

https://github.com/espruino/Espruino/assets/3240233/8f0c2ec8-25e2-47fe-9316-a5668f664ed4

I think it hasn't always been hidden away there, I remember seeing those reinstall arrows by default. I looked for them when I needed them because of the settings app, and could not find them, and gave up for a few days, thinking they had been removed ... until I discovered that new tiny 'Advanced' there at the bottom.

Moini, 2023-10-02:
(the reset button is placed in a typical 'Save settings' position. A bit dangerous! I almost clicked it.)

gfwilliams, 2023-10-02:
Ahh, ok - thanks! I'm pretty sure that's a new thing that didn't exist before. You could however always just start the app loader and click 'install' next to an app before you'd connected previously and that should do it.