espruino/Espruino #2424: Division of Non-Numeral Operands Triggers Syntax Error on Linux

marccarigiet, 2023-10-30:
- Plattform: [Linux, Ubuntu 20.04]

**Description**
Certain divisions of non-numeral operands get interpreted as an unfinished Regex and trigger a Syntax Error.

**Test case**
Test code to reproduce the behaviour:
```js
true / false;

```

**Backtrace**
```
Uncaught SyntaxError: Got UNFINISHED REGEX expected EOF
 at line 1 col 6
true / false;
```

**Expected behaviour**
I am not entirely sure what the expected behaviour should be. Going by other JavaScript Engines, NaN or Infinity would both be possible. But I definitely wouldn't have expected the present behavior.

**Additional context**
This was found using comparative testing between multiple JavaScript engines. This could very much come down to a design choice. I just wanted to let you know in case it is not intentional

gfwilliams, 2023-11-02:
Thanks - I've just put a fix in for this. JS Regex is nasty as it's not apparent to the poor lexer (without lookahead) exactly what makes a RegEx start and what makes a division.

Do you have any other examples of non-numerals beforehand that cause issues that work on other engines? I fixed `true/false/undefined/null`

I'd be particularly interested if there was everyday JS that had problems - I feel like dividing `true` by something is probably not something that would happen in non-buggy code :)

closed 2023-11-02 by gfwilliams

marccarigiet, 2023-11-02:
Hi there, 

The only other non-numerals that work on other engines and caused that issue were with this and uninitialized variables. Admittedly divisions involving theses are also not everyday JS 😅

commit 0f1b5b00 (2023-11-02): Fix for UNFINISHED REGEX syntax error when parsing `true / false` (fix #2424)