espruino/Espruino #2426: new Uint8Array -> reset

mikabytes, 2023-11-06:
Previously I was on version 2.14 and all was fine. But with 2.18 and 2.19 I have this error:

```
>new Uint8Array(3)
```

```
=F 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 80MHz
  SPI Mode       : DIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

nÇrn
    l`rlÄlþLoading 5717 bytes from flash...
```

gfwilliams, 2023-11-06:
This works in the normal build, so what hardware are you running this on? And is it a build we supply or one you did yourself?

mikabytes, 2023-11-06:
This is running on an ESP8266 with firmware downloaded from the releases page ([here to be specific](https://www.espruino.com/binaries/espruino_2v19_esp8266_4mb/))

mikabytes, 2023-11-06:
Also tried on a completely fresh install, no JS loaded whatsoever. Just cli using Espruino binary from NPM

gfwilliams, 2023-11-06:
Ok, thanks!

> Also tried on a completely fresh install,

Are you sure? Because it says `Loading 5717 bytes from flash...` like you might have some code already in there which might be causing some kind of issue? You could try running `reset(1)` which should hopefully get rid of that

mikabytes, 2023-11-06:
Yes, initial bug report was indeed on a chip that had other JS loaded on it. Here's the output after running `reset(1)`

```
>reset(1)
Erasing saved code.
 ____                 _ 
|  __|___ ___ ___ _ _|_|___ ___ 
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v19 (c) 2021 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xd8 chip 0x4016

>new Uint8Array(3)
=new 
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 80MHz
  SPI Mode       : DIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000

nÇrn
    l`rlÄlþ
 ____                 _ 
|  __|___ ___ ___ _ _|_|___ ___ 
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v19 (c) 2021 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xd8 chip 0x4016
```

gfwilliams, 2023-11-06:
Ok, thanks - it's interesting this one lacks the `F 0` - I'll try and take a look when I get a second. I know ESP8266 is very picky about accessing non-byte aligned RAM areas and it's possible something's crept in which causes a problem

mikabytes, 2023-11-06:
Another interesting observation; it will allow me to create signed integers *once*, but if I do it *twice* then it will reset.

```
>reset(1)
Erasing saved code.
 ____                 _ 
|  __|___ ___ ___ _ _|_|___ ___ 
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v19 (c) 2021 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xd8 chip 0x4016

>new Int8Array(3)
=new Int8Array([111, 114, 121])
>new Int8Array(3)
=F 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 80MHz
  SPI Mode       : DIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000
```

gfwilliams, 2024-01-31:
This appears to be fixed now - I'm pretty sure it was due to an overflow in the JsVar union for arraybuffers, which only showed up on devices with little RAM

closed 2024-01-31 by gfwilliams

mikabytes, 2024-02-01:
Great, thanks!

I can confirm. Tested my old code on v2.20 and it's running fine now.