espruino/Espruino #2439: Uint8Array constructor byteOffset is restricted to 2bytes

wagnerf42, 2023-12-08:
hi,

byteOffset in the Uint8Array constructor stops working when the integer goes over 65536.
it just wraps around so surely this offset is coded in C as a 2 bytes integer.

i'm not sure how feasible it would be to switch to a larger integer ?

wagnerf42, 2023-12-08:
thinking again about it, it's not such a big deal, I will be able to get around it using storage.read(). i'm not even sure we did'nt already discuss this actually.
so, maybe we could close the issue.

gfwilliams, 2023-12-08:
Ok, well, I think this isn't something easily fixable - we only have a certain amount of space in the structure to store an offset.

However it should throw an error, not wrap - i'll see about getting a fix in

commit 1e697ff2 (2023-12-08): Ensure we error if an unsupported byteOffset is supplied to types array constructor (fix #2439)
closed 2023-12-08 by gfwilliams