espruino/Espruino #2440: Does the build_board_docs.py script currently work

brendena, 2023-12-12:
I tried to build the documentation with EspruinoDocs/build.sh, but there were errors in the build_board_docs.py in this repo.    
The error i saw was that it was throwing a error [here](https://github.com/espruino/Espruino/blob/master/scripts/common.py#L257).  It would fail if the array didn't have a zero's index.  Which happened over the dicky watch and possibly others json object parsing.  

My goal is to see the module page i'm working on.  So i tried to[ fix it here ](https://github.com/espruino/Espruino/compare/master...brendena:Espruino:Python3_build_docs) by moving to **Python3** and fixing the errors i saw.  But the output looked wrong.  My best guess is bootstrap is not loading which has many of the classes needed.  Lower down the page there are css classes that do get loaded.  

[image]

gfwilliams, 2023-12-13:
Thanks for trying to check your module's rendering out beforehand!

That's odd - I do see it working here - I use it to build the website.

So what call do you see an error on? I just tried on:

```
python scripts/build_board_docs.py ESPRUINOBOARD pinout
```

And it appears to work fine for me. Your changes to get it working with Python 3 sound like a good idea either way though.

As for the result - I think it's correct? Comparing with the printable version at https://www.espruino.com/Original?print it looks about right? It would look a bit different to what it does on the main site, but we don't generally use bootstrap in the reference pages...

gfwilliams, 2023-12-13:
Thanks!

But `Dicky JS`? I thought that was a typo. Where are the files for that board?

brendena, 2023-12-13:
My bad it was Dickens.  So this is the message i would get from running **EspruinoDocs/build.sh**

It wouldn't crash the build.  I just didn't know what other ramifications it had because it failed.  Since when it its this error it does a exit(1).

[This is the error i would get]([url](https://github.com/espruino/Espruino/blob/master/scripts/common.py#L257))

JSON PARSE FAILED for {
    "type" : "staticmethod", "class" : "Bangle", "name" : "drawWidgets", "patch":true,
    "generate_js" : "libs/js/dickens/Bangle_drawWidgets_DICKENS.min.js",
    "#if" : "defined(BANGLEJS) && defined(DICKENS)"
} - list index out of range
Traceback (most recent call last):
  File "scripts/build_docs.py", line 49, in <module>
    jsondatas = common.get_jsondata(True)
  File "/home/brendena/Espruino/scripts/common.py", line 271, in get_jsondata
    print(''.join(traceback.format_exception(None, exc_obj, exc_obj.__traceback__)))
NameError: global name 'traceback' is not defined

gfwilliams, 2023-12-14:
Thanks! That's really odd - I wonder why I didn't see it in my builds.

As far as I can tell I'm executing it using Python 3.10.12 so it's not like I'm using Python 2.7. Maybe in a newer release they made it complain about accessing `[0]` on an empty array - thanks for fixing that code, as it looked broken anyway

closed 2023-12-14 by gfwilliams