espruino/Espruino #2446: Stopping code execution when the terminal is not connected

i-terezie, 2024-01-13:
I have carefully studied the old correspondence on the problem of software shutdown due to the suspension of exchange via the console. And I want to note its importance and the lack of a solution to this critical problem.
To begin with, I will note two important points that were discussed earlier. The first is the importance of delivering the information sent. And the second is the importance of maintaining the main code's operability. These are two contradictory aspects. But at the same time, both of them are critically important for solving engineering problems.
The position of the developers that data delivery is important is certainly correct, but at the same time there is also a second aspect, it is the promptness of information delivery. I'll give you an example. There is a smart home controller that informs the control computer about pressing buttons, temperature changes… And it is important for him to monitor the temperature, press buttons, turn on/off the light, and so on. But the loss of information when transferring it to the  computer does not bother him. But at the same time, information should always be sent quickly. After an hour, no one needs it anymore, and a hung controller is generally bad. This is how TCP/IP and UDP packets work. The first ones are lossless, the second ones are operational.
The ability to organize sending data to the console, similar to UDP, should be available, but it is not. As a solution, I can offer two options.
1) A special mode is enabled for the console, in which data is thrown out if it is impossible to send.
2) A more intelligent one, in which data that has not been sent, for any reason, is thrown away within a given time.

I note that this should not be the default behavior, namely the behavior for special cases, through an additional call. For many tasks, this will become the main mode of operation. For many, it won't, but it should be!

If this mode has already been implemented, then I will be glad to use it in my project

gfwilliams, 2024-01-15:
Thanks - out of interest, what device are you using where you're encountering an issue?

For Bluetooth devices we took the view eventually that we'd just throw data out, so perhaps it's time to consider that for USB as well.

gfwilliams, 2024-02-01:
Just fixed

commit d217500f (2024-02-01): STM32: (Original/Pico/WiFi) if USB connected but PC not receiving data, throw away USB data rather than blocking (fix #2446)
closed 2024-02-01 by gfwilliams