espruino/Espruino #2451: Constructor comparisons don't work

YekkeTechie, 2024-01-19:
Regarding primitive types, comparisons like `<T literal>.constructor == <T>.prototype.constructor` do not always give consistent results. For example
```js
const cmp = () => "blah".constructor == String.prototype.constructor;
console.log(
    cmp() == cmp()
);
```
prints
```
>false
```
<br/>

(I'm using version `2v18.1`)

gfwilliams, 2024-01-19:
Thanks - interestingly, `"blah".constructor` seems to return `undefined` initially, but if you call `String.prototype.constructor` it's fine afterwards - so it seems to be having trouble creating the constructor.

So if you change the order of comparison, it appears to work reliably for me? Is that the same for you?

```
const cmp = () => String.prototype.constructor == "blah".constructor;
console.log(
    cmp() == cmp()
);
```

gfwilliams, 2024-01-19:
I've got a fix for this, but it appears to break some other stuff so I'm going to hold off committing it.

I believe the 'real prototype chain' branch would fix this too.

commit ddd7ad7e (2024-01-24): Fix issue requesting constructor of builtin before the class itself has been accessed (fix #2451)
closed 2024-01-24 by gfwilliams