espruino/Espruino #2456: SEGV on unknown address in jsvStringIteratorPrintfCallback

Ye0nny, 2024-01-26:
###### Espruino revision
Commit: https://github.com/espruino/Espruino/commit/fcc9ba4f12d79710f93d47ea37987cde0efce410
Version: 2v20

###### Build platform
Ubuntu 20.04.5 LTS (Linux 5.4.0-144-generic x86_64)

###### Build steps
```sh
export CCFLAGS='-g -fsanitize=address -fno-omit-frame-pointer'
export LDFLAGS='-fsanitize=address'
DEBUG=1 make
```

###### Test case
<details><summary>testcase</summary>
<p>

```javascript
d = new Date ( - 2147483648 , 32 ) ; 
if ( " Invalid Date " != d . toString ( ) ) throw ' Expected " Invalid Date " , but got : " ' + d + ' " ' ;
```
</p>
</details>

```javascript
// poc.js
d = new Date ( - 2147483648 , 32 ) ; 
d . toString ( )
```

###### Execution steps & Output
```
$ ./jerryscript/build/bin/jerry poc.js

 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v20 (c) 2023 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate

AddressSanitizer:DEADLYSIGNAL
=================================================================
==1936175==ERROR: AddressSanitizer: SEGV on unknown address 0x5595cc335f38 (pc 0x5595cf7bb33e bp 0x7fffd42959f0 sp 0x7fffd42959e0 T0)
==1936175==The signal is caused by a READ memory access.
#0 0x5595cf7bb33d in jsvStringIteratorPrintfCallback src/jsvar.c:1864
#1 0x5595cf7c94c1 in vcbprintf src/jsutils.c:818
#2 0x5595cf7bb570 in jsvVarPrintf src/jsvar.c:1890
#3 0x5595cf832863 in jswrap_date_toString src/jswrap_date.c:699
#4 0x5595cf7ca79e in jsnCallFunction src/jsnative.c:222
#5 0x5595cf7cc85e in jspeFunctionCall src/jsparse.c:656
#6 0x5595cf7ce45a in jspeFactorFunctionCall src/jsparse.c:1322
#7 0x5595cf7d0753 in jspePostfixExpression src/jsparse.c:1946
#8 0x5595cf7d08d2 in jspeUnaryExpression src/jsparse.c:1970
#9 0x5595cf7d119e in jspeBinaryExpression src/jsparse.c:2129
#10 0x5595cf7d137d in jspeConditionalExpression src/jsparse.c:2165
#11 0x5595cf7d16ea in jspeAssignmentExpression src/jsparse.c:2225
#12 0x5595cf7d170c in jspeExpression src/jsparse.c:2231
#13 0x5595cf7d4029 in jspeStatement src/jsparse.c:3019
#14 0x5595cf7d1a72 in jspeBlockOrStatement src/jsparse.c:2331
#15 0x5595cf7d1aa8 in jspParse src/jsparse.c:2342
#16 0x5595cf7d4b1b in jspEvaluateVar src/jsparse.c:3275
#17 0x5595cf7d4c99 in jspEvaluate src/jsparse.c:3305
#18 0x5595cf82b0bb in main targets/linux/main.c:512
#19 0x7f8d9482d082 in __libc_start_main ../csu/libc-start.c:308
#20 0x5595cf7b1e0d in _start (./espruino/bin/espruino+0x1ae0d)

 AddressSanitizer can not provide additional info.
 SUMMARY: AddressSanitizer: SEGV src/jsvar.c:1864 in jsvStringIteratorPrintfCallback
 ==1936175==ABORTING

```
---
when executed in release mode

**Output**
```

 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v20 (c) 2023 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate

Segmentation fault
```

Credits: @Ye0nny, @EJueon

gfwilliams, 2024-01-26:
Thanks! Looking into it now, on the [DST PR](https://github.com/espruino/Espruino/pull/2230) that seemed to cause this issue

gfwilliams, 2024-01-29:
Fixed with #2459

closed 2024-01-29 by gfwilliams

Ye0nny, 2024-01-30:
Thanks for fixing :)