espruino/Espruino #2470: Bangle.js1 - "< Back" duplicates

Erovia, 2024-02-26:
According to [the Graphical Menu](https://www.espruino.com/graphical_menu) documentation, each menu can have a `"back"` key in the `""` section.  
My understanding is that this could/should be preferred over explicitly creating menu options for moving between menus.

However, on the Bangle.js 1 this only works as intended when the user opens the submenu for the first time.  
On the second (and following) entries there will be two `< Back` options in the menu.  
One at the top, one at the bottom.  
[image]

This is because after entering the menu for the first time, a `"< Back"` key is created in the menu object, outside of `""`.

## Repro

1) Execute test code on the Bangle.js 1 emulator:

```js
var mainmenu = {
  "" : {
    "title" : "-- Main Menu --"
  },
  "Submenu" : function() { E.showMenu(submenu); },
};

var submenu = {
  "" : {
    "title" : "-- SubMenu --",
    "back": function() { E.showMenu(mainmenu); }
  },
  "test": "",
};

E.showMenu(mainmenu);
```

2) State of submenu:

```js
>submenu
={
  "": {
    title: "-- SubMenu --",
    back: function () { ... }
   },
  test: ""
 }
```

3) Entering the submenu:

```js
>E.showMenu(submenu)
={ lastIdx: 0,
  draw: function (a,b) { ... },
  select: function () { ... },
  move: function (a) { ... },
  back: function () { ... }
 }
>
>submenu
={
  "": {
    title: "-- SubMenu --",
    back: function () { ... },
    fontHeight: 16, selected: 0 },
  test: "",
  "< Back": function () { ... }
 }
```
At this point however, the menu renders correctly with only one `< Back` option.

4) Leave the submenu and enter it again:

```js
>E.showMenu(mainmenu)
={ lastIdx: 0,
  draw: function (a,b) { ... },
  select: function () { ... },
  move: function (a) { ... },
  back: function () { ... }
 }
>
>E.showMenu(submenu)
={ lastIdx: 0,
  draw: function (a,b) { ... },
  select: function () { ... },
  move: function (a) { ... },
  back: function () { ... }
 }
```

This time, there will be two `< Back` options, one at the top and one at the bottom, both working.

gfwilliams, 2024-02-26:
Thanks! That's a pain - I'm surprised this hasn't come up before as it will have been like this for a while, but maybe for the majority of uses E.showMenu gets called with a newly created argument each time so it's not obvious.

Have you seen this on a real device, or is it emulator only?

Erovia, 2024-02-26:
This is reproducible on real hardware running 2v21 with all apps updated.

commit 43f806d6 (2024-05-21): Bangle.js1: Fix issue where going out of and back into a menu would cause 'Back' to be added twice (fix #2470)
closed 2024-05-21 by gfwilliams