espruino/Espruino #2483: Build LINUX board on Mac Os X Intel

MaBecker, 2024-04-11:
Hi, wanted to know if this is Mac gcc specific or if there is someting broken when building a Espruino on Mac as Linux board.

```c
#/usr/bin/gcc -v
# Apple clang version 15.0.0 (clang-1500.3.9.4)
# Target: x86_64-apple-darwin23.4.0

targets/linux/jshardware.c:361:48: error: incompatible function pointer types passing 'void (*)()' to parameter of type 'void * _Nullable (* _Nonnull)(void * _Nullable)' [-Wincompatible-function-pointer-types]
  int err = pthread_create(&inputThread, NULL, &jshInputThread, NULL);
                                               ^~~~~~~~~~~~~~~

```

MaBecker, 2024-04-11:
The last time I tried to build a Linux board was release 2v18.6.

gfwilliams, 2024-04-11:
Pretty sure this is Mac specific - works fine on Linux

opichals, 2024-04-12:
@MaBecker Seeing the same error here on M1.

```sh
% gcc -v
Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: arm64-apple-darwin23.4.0
```

```c
targets/linux/jshardware.c:361:48: error: incompatible function pointer types passing 'void (*)()' to parameter of type 'void * _Nullable (* _Nonnull)(void * _Nullable)' [-Wincompatible-function-pointer-types]
  int err = pthread_create(&inputThread, NULL, &jshInputThread, NULL);
                                               ^~~~~~~~~~~~~~~
```

MaBecker, 2024-04-12:
Found [this](https://stackoverflow.com/questions/58087929/incompatible-pointer-types-passing-void-void-to-parameter-of-type-void), adding it is building but causing a lot linker warnings like this:

```c
53 warnings generated.
LD bin/espruino
ld: warning: disabling chained fixups because of unaligned pointers
ld: warning: pointer not aligned at _jswSymbols_Pin_proto+0x4 from  /github/Espruino/Espruino/obj/gen/jswrapper.o
.....
looks like for al _jswSysmbols_*
.....
```

I will keep this localy until someone comes up with a propper solution for this on Mac's.

MaBecker, 2024-04-15:
> Pretty sure this is Mac specific - works fine on Linux

Well, I guess it's more a gcc version thing ;-)

opichals, 2024-10-28:
Is this still an issue?

I seem not to be getting any ld warnings with the following setup:

```patch
% git diff    
diff --git a/targets/linux/jshardware.c b/targets/linux/jshardware.c
index 8475a183a..2494d769c 100644
--- a/targets/linux/jshardware.c
+++ b/targets/linux/jshardware.c
@@ -252,7 +252,7 @@ int getch() 
-void jshInputThread() {
+void* jshInputThread() {
   while (isInitialised) {
```

```
% sw_vers                 
ProductName:		macOS
ProductVersion:		15.0.1
BuildVersion:		24A348
```
```
% gcc --version
Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: arm64-apple-darwin24.0.0
Thread model: posix
```
```
% ld -v
@(#)PROGRAM:ld PROJECT:ld-1053.12
BUILD 15:45:29 Feb  3 2024
configured to support archs: armv6 armv7 armv7s arm64 arm64e arm64_32 i386 x86_64 x86_64h armv6m armv7k armv7m armv7em
will use ld-classic for: armv6 armv7 armv7s arm64_32 i386 armv6m armv7k armv7m armv7em
LTO support using: LLVM version 15.0.0 (static support for 29, runtime is 29)
TAPI support using: Apple TAPI version 15.0.0 (tapi-1500.3.2.2)
```

opichals, 2024-10-28:
Unsure about the Intel target though. Perhaps something like this could help?
```
% git diff
diff --git a/src/jswrapper.h b/src/jswrapper.h
index 54b85bc92..9e20df940 100644
--- a/src/jswrapper.h
+++ b/src/jswrapper.h
@@ -67,7 +67,7 @@ typedef enum {
 // reads. Telling the compiler to pack the structs defeats that, so we have to take it out.
 #define PACKED_JSW_SYM
 #endif
-#if defined(__arm64__)
+#if defined(__x86_64__) || defined(__arm64__)
 #undef PACKED_JSW_SYM
 #define PACKED_JSW_SYM __attribute__((aligned(2)))
 #endif
```

looran, 2025-01-09:
I get the same compile error with GCC 14, but compilation passes on GCC 13, as [suggested above](https://github.com/espruino/Espruino/issues/2483#issuecomment-2055763474).

To compile specifically with GCC 13 I use:
`CCPREFIX=/usr/x86_64-pc-linux-gnu/gcc-bin/13/ make`

gfwilliams, 2025-01-09:
Thanks! Does this change fix it for you with GCC 14?

```
% git diff    
diff --git a/targets/linux/jshardware.c b/targets/linux/jshardware.c
index 8475a183a..2494d769c 100644
--- a/targets/linux/jshardware.c
+++ b/targets/linux/jshardware.c
@@ -252,7 +252,7 @@ int getch() 
-void jshInputThread() {
+void* jshInputThread() {
   while (isInitialised) {
```

looran, 2025-01-09:
Surely does, compilation passes with your patch on gcc `14.2.1`

gfwilliams, 2025-01-09:
Great, thanks! Sorry this took so long to get fixed but I can't test here - just pushed @opichals change in

commit d55ac847 (2025-01-09): Fix Native build on MacOS w GCC 14 thanks to @opichals patch (fix #2483)
closed 2025-01-09 by gfwilliams