espruino/Espruino #2484: Storage.writeJSON and Storage.readJSON are incompatible when numeric keys are present

PVTejas, 2024-04-14:
[espruino/Espruino@59dda13](https://github.com/espruino/Espruino/commit/59dda13c259888671276a55dbcdb7579223d6329) breaks `Storage.readJSON` in cases where dict keys are purely numeric

```
require("Storage").writeJSON("test",{"1":1});
require("Storage").readJSON("test");
```

Throws the error `Got INT expected '}'`

Related issues:
- https://github.com/espruino/BangleApps/issues/3310
- https://github.com/espruino/BangleApps/issues/3346

halemmerich, 2024-04-14:
It could be later commits on the same code that break this, https://github.com/espruino/Espruino/commit/59dda13c259888671276a55dbcdb7579223d6329 is just the first introducing the change.

PVTejas, 2024-04-15:
That is likely the case. My bad.

gfwilliams, 2024-04-15:
Thanks! Just fixed it hopefully so on new firmwares this should all work again

commit b251d831 (2024-04-15): Storage.readJSON will now read numeric field names like "{1:1}" which can be produced by writeJSON (fix #2484)
closed 2024-04-15 by gfwilliams