espruino/Espruino #2487: Wifi.connect(ssid, options, callback) does not handle callback

MaBecker, 2024-04-19:
here my snippet for testing the call back

```js
  wifi.connect(SSID, {password:SSIDPASS}, function(err){
    console.log("connected? err=", err, "info=", wifi.getIP());
  });
```

no callback is fired .....

https://github.com/espruino/Espruino/blob/fefa704fb920b1ac902ebe8e50fb202c36e83c17/libs/network/esp32/jswrap_esp32_network.c#L776-L895

Any hints why it's not called at all?

gfwilliams, 2024-04-19:
Well I just flashed the latest firmware onto an ESP32, ran that exact code, and it works great for me. So the question I guess is what's different in your case? Do you have a custom build? if your Wifi network strange in some way?

Presumably this worked in the past for you? You would have noticed surely?

Also I'd imagine a lot more people would complain if that function was completely broken

MaBecker, 2024-04-19:
```js
var wifi=require("Wifi");
wifi.on("associated", function(details) {
	console.log("Event-> associated: " + JSON.stringify(details));
});
wifi.on("connected", function(details) {
	console.log("Event-> connected: " + JSON.stringify(details));
});
wifi.on("disconnected", function(details) {
	console.log("Event-> disconnected: " + JSON.stringify(details));
});
wifi.on("auth_change", function(details) {
	console.log("Event-> auth_change: " + JSON.stringify(details));
});
wifi.on("dhcp_timeout", function(details) {
	console.log("Event-> dhcp_timeout: " + JSON.stringify(details));
});
wifi.on("probe_recv", function(details) {
	console.log("Event-> probe_recv: " + JSON.stringify(details));
});
wifi.on("sta_joined", function(details) {
	console.log("Event-> sta_joined: " + JSON.stringify(details));
});
wifi.on("sta_left", function(details) {
	console.log("Event-> sta_left: " + JSON.stringify(details));
});
wifi.connect(ssid, {password: password}, function() {
	console.log("Connect completed");
});

wifi.connect(ssid, {password:password}, (d)=>{
    print("callback wifi.connect, d:", d);
});

// output
/*
Event-> associated: {"ssid":"XXXXXXXX","mac":"84:db:ac:43:b0:5f","channel":"1"}
callback wifi.connect, d: null
Event-> connected: {"ip":"192.168.101.100","netmask":"255.255.255.0","gw":"192.168.101.1"}
*/
```

looks all fine, just picked the wrong test coding.

closed 2024-04-19 by MaBecker