espruino/Espruino #2490: Debugger feature: add backtrace command

bobrippling, 2024-04-21:
I'd like to be able to see the call stack (useful for debugging what's called code, such as finding out where a [`width` has been set](https://github.com/espruino/BangleApps/pull/3320#issuecomment-2062244090)), like `bt` in `gdb` - before I work on the feature though, is this something that would be accepted into espruino?

I'm thinking something like:
```
debug>bt
Frame 0: drawIcon(), "clock.js", line 5
Frame 1: draw(), "clock.js", line 20
Frame 2: anonymous, "clock.js", ...
```

gfwilliams, 2024-04-22:
Well, if this were possible in an efficient way I'd be all for it. However if it were easy I'd have added it to the debugger already :)

Basically I don't want to calculate/store a bunch of extra stuff for each function call, because that would slow execution down.

So I guess you have two options really:

* Actually skip back down the stack (this will be platform dependent) a bit like GDB's `bt` would do, and find instances of `jspeFunctionCall`
* Add an `oldLex` pointer to JsLex, and as we replace it when we're executing the function (https://github.com/espruino/Espruino/blob/176a23cf1bf8e37aab69b125b316942bced8332e/src/jsparse.c#L840) link in the 'old' lexer rather than storing it in `JsLex *oldLex`. That way the debugger could skip back down all the lexers and reconstruct the call stack.

Now I think about it the second option should be possible with minimal extra overhead

bobrippling, 2024-04-22:
Interesting - I'll take a look, thanks for the pointers

closed 2025-05-23 by gfwilliams

gfwilliams, 2025-05-23:
I've just fixed this. It actually worked really nicely linking back on the stack, and meant I could avoid some of the previous hackiness for stack traces. For instance you can now do `(new Error()).stack` and get a proper stack trace in your code at the point you call it. `console.trace()` also displays a stack trace too.

I've taken the opportunity to allow the stack trace to tag previously untraceable stuff too (for example when running code from `.boot0` or the REPL), and I fixed the `Uncaught undefined` issue if there was a bug in `app.boot.js`

bobrippling, 2025-06-03:
Great, thanks!

commit 15923905 (2025-05-23): Add a backtrace command to debugger, add console.trace command, set Error.stack in constructor (fix #2490)
            Error.stack/stack dump now uses more standard file:line:col format
            Out of memory errors now show a backtrace (previously it was very hard to track these down)