espruino/Espruino #2492: E.getPowerUsage to estimate power consumption

gfwilliams, 2024-04-22:
We don't know exactly how much power is being used by certain things, but on some devices we do have an idea, and we could expose that to the user.

* Check the bluetooth connection state / connection / advertising interval and guess on BLE power usage
* Compare SYSTICK (system clock ticks) against RTC to figure out how many cycles we executed (on nRF52 it's roughly 4mA if running all the time)
* Are LEDs on? 8mA per LED/etc
* On Puck.js/Bangle/etc we will know if certain peripherals are powered on and roughly how much they draw

So it could report:

```
>E.getPowerUsage()
={
 usage : 5.5, // mA
 d : [
  code : 2,
  BLE : 1,
  accel : 0.5,
  LCD : 2,
  // ...
 ]
}
```

We could implement something using `build_jswrapper.py` so it's easy to extend for different devices.

Any thoughts? It feels like a good feature as users often don't have a way to easily check power usage.

halemmerich, 2024-04-23:
I like this, it would reduce the code of the [`powermanager`](https://github.com/espruino/BangleApps/blob/master/apps/powermanager/widget.js) widget by a lot.
Backlight power use on the bangles depending on current brightness would be nice too as it is a huge draw.

gfwilliams, 2024-04-24:
Fixed with fa73e3103165573bf18f75a8e40e46dee6189ccc

closed 2024-04-24 by gfwilliams