espruino/Espruino #2499: ESP32_IDF4

MaBecker, 2024-05-15:
This build is on development stage, collecting issues here.

c3c, 2024-05-20:
Hey, I realize this is in active development, just eager to test on this new board and wanted to report what currently works for me.

I compiled for ESP32C3_IDF4 and uploaded the firmware. So far I'm able to connect via BLE and get a working REPL. 
Via the Web IDE (over BLE) the device disconnects as soon as I try to upload data/files.
Via USB serial I don't seem to get 2-way interaction (although I do see the bootloader printouts and espruino logo). I'm guessing that might have to do with a slightly different pinout(?) Using M5StampC3.

Is this expected with what is currently implemented?

gfwilliams, 2024-05-21:
> Via USB serial I don't seem to get 2-way interaction

What happens if you press enter? Just yesterday I fixed the issue where the REPL didn't update immediately (only after newline) so that could be what you were seeing?

c3c, 2024-05-22:
I had cloned the Git commit right before your REPL commit. Looks like I have the same issue though on master [e1838a2](https://github.com/espruino/Espruino/commit/e1838a26c705e92aaf5e1e8e78e8ec3ed7233c91) (I made sure to `make clean` just in case).

Can the `default_console` configuration affect this?

gfwilliams, 2024-05-22:
I don't think `default_console` would affect it - I guess it could even be your terminal app (if it's got RTS/CTS flow control enabled)

gfwilliams, 2024-10-10:
All items complete now and it's available as a cutting edge build, closing this. Definitely feels like a usable build now

closed 2024-10-10 by gfwilliams