espruino/Espruino #2500: Any plans to switch to python3?

MaBecker, 2024-05-18:
Some scripts do still user `python`instead of `$PYTHON`. For the make files it is already implemented. Any plans to switch?

gfwilliams, 2024-05-20:
I think all the scripts are compatible with python3 now. Which ones did you find that don't reference it directly?

I'd rather not use $PYTHON - it's not set on my system anyway, and IMO that's just a worse way of writing `python`. We should just use `python3`

MaBecker, 2024-05-21:
this link will give you a list 

https://github.com/search?q=repo%3Aespruino%2FEspruino%20%2Fusr%2Fbin%2Fenv%20python&type=code

gfwilliams, 2024-05-21:
If you want to change them, check they're ok and do a PR that'd be great? But from my point of view this isn't a high priority as it all works for me as-is

MaBecker, 2024-05-21:
look's like you have linked python3 to python or somethin else to make python work ;-)

gfwilliams, 2024-05-21:
`sudo apt install python-is-python3`

I don't believe I have done anything special though - all the GitHub action builds work fine as is

MaBecker, 2024-09-07:
Ok,  it's all Mac OS X based, so let's close it.

closed 2024-09-07 by MaBecker