espruino/Espruino #2502: `Object.clone()` returning function

bobrippling, 2024-05-18:
Could I be using it wrong? I'd like to have a deep copy of an object, which might contain further objects or arrays

```javascript
>c = Object.clone({ a: 1 })
=function () { [native code] }
>c.a
=undefined
```

gfwilliams, 2024-05-20:
Ahh - yes, sorry - it's meant to be called on the Object itself (although I admit your usage makes more sense!). I'll update the documentation with an example. However is it *not* a deep copy I'm afraid:

```
a = { d : [1,2,3] };
b = (a).clone()
>b
={ d: [ 1, 2, 3 ] }
```

For that you're probably back to `JSON.parse(JSON.stringify(x))` which is pretty nasty.

As far as I can tell the function is basically the same as `Object.assign({}, obj)` - I'm actually kind of tempted to remove it completely. I think in the very early days before we had `Object.assign` it was useful, but now it's really not.

closed 2024-05-20 by gfwilliams

bobrippling, 2024-05-21:
Ahhh doh, yes that was rather silly of me. That's great, thank you