espruino/Espruino #2505: Using different pins for neopixel on ESP32 does not work correct

MaBecker, 2024-05-24:
Test Code:

```js
np = require("neopixel");
// set first five neopixel blue on pin D16 - ok
np.write(D16,[0,0,40, 0,0,40, 0,0,40, 0,0,40, 0,0,40]);
// set first  five neopixel green on pin D17 - changes pixel on D16 to green too - err
np.write(D17,[40,0,0, 40,0,0, 40,0,0, 40,0,0, 40,0,0])
```

The neopixel implementation does not detach pins which is causing this effect. 

First approche to fix this:

```c
int neopixelConfiguredGPIO = -1;
....
void neopixel_init(int gpioNum){
....
  if (neopixelConfiguredGPIO != gpioNum) {
    // detach last pin 
    if (neopixelConfiguredGPIO)
      gpio_matrix_out(neopixelConfiguredGPIO,0,0,0);
    neopixelConfiguredGPIO = gpioNum;
  }    
  rmt_set_pin((rmt_channel_t)RMTCHANNEL, RMT_MODE_TX, (gpio_num_t)gpioNum);
....
}
```

Same behavior for all ESP-IDF versions. Needs more testing before sending a pr.

MaBecker, 2024-08-21:
This approach does not work reliable. Better use `Pin.mode("input") after write call. Will send a pm  after some more testing.

rgomezwap, 2024-08-21:
Hello Mark, I've modified the BUILD system so we can generate binaries for ESP32, ESP32C3, and ESP32S3 using ESP-IDF 4.4.8. I've added WiFi, Bluetooth, and Neopixel support. However, there are no guarantees that it will work.

On the ESP32-S3, I quickly tested Neopixel, but it's not working for me. So, it's true that the Neopixel functionality on ESP is something that needs to be thoroughly checked and tested.

MaBecker, 2024-08-22:
Did you tested neopixel with multiple stips? When I tested all strips showed the same. All samples from Espressif only show samples for one pin......

https://components.espressif.com/components/espressif/led_strip

MaBecker, 2024-08-23:
ESP32 & C3 use  `gpio_matrix_out(neopixelConfiguredGPIO, 0x100, 0, 0);`  and it is stable again.

rgomezwap, 2024-08-23:
That's great news. Once it's updated in the master branch, I'll run a test with the ESP32-S3. I believe it will work correctly as well, let's hope so.

closed 2024-08-27 by MaBecker