espruino/Espruino #2509: banglejs_iflash build: compacting corrupts `.boot0`, error running `.bootcde`

thyttan, 2024-06-03:
I roughly did this:
1. Installed Cutting Edge build: espruino_2v22.8_banglejs2_iflash.zip from app loader
2. factory reset my watch
3. removed all apps with the app loader button under the "More..." tab
4. auto set up my watch (around 40 apps) by installing from my app loader: https://thyttan.github.io/BangleApps/?q=mysetup
5. uninstall `fastreset`
6. Install from my app loader [`fastbtnload`](https://thyttan.github.io/BangleApps/?q=shortbtnload) after the following happened:

```
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v22.8 (c) 2024 G.Williams
>
{"t":"act","stp":12,"hrm":0,"mov":248}
{"t":"status","bat":81,"chg":0}
{"t":"act","stp":30,"hrm":0,"mov":254}
[{"id":"sched","version":"0.26","files":"sched.info,sched.boot.js,sched.js,sched.img,sched,sched.settings.js,sched.clkinfo.js","data":"sched.json,sched.settings.json","type":"scheduler"},{"id":"kbmulti","version":"0.09","files":"kbmulti.info,textinput,kbmulti.settings.js","data":"kbmulti.settings.json","type":"textinput"},{"id":"messageicons","version":"0.07","files":"messageicons.info,messageicons,messageicons.img","type":"module"},{"id":"messagegui","version":"0.79","files":"messagegui.info,messagegui,messagegui.app.js,messagegui.new.js,messagegui.boot.js,messagegui.img"},{"id":"widmessages","version":"0.06","files":"widmessages.info,widmessages.wid.js","type":"widget"},{"id":"messages","version":"0.62","files":"messages.info,messages,messages.settings.js","data":"messages.json,messages.settings.json","type":"module"},{"id":"widmsggrid","version":"0.06","files":"widmsggrid.info,widmsggrid.wid.js","type":"widget"},{"id":"msgwakefup","version":"0.01","files":"msgwakefup.info,msgwakefup.boot.js","type":"bootloader"},{"id":"notify","version":"0.14","files":"notify.info,notify","type":"notify"},{"id":"health","version":"0.30","files":"health.info,health.app.js,health.img,health.boot.js,health,health.settings.js","data":"health.json"},{"id":"widminbate","version":"0.05","files":"widminbate.info,widminbate.wid.js","type":"widget"},{"id":"podadrem","version":"0.10","files":"podadrem.info,podadrem.app.js,podadrem.img"},{"id":"spotrem","version":"0.11","files":"spotrem.info,spotrem.app.js,spotrem.img"},{"id":"android","version":"0.36","files":"android.info,android.app.js,android.settings.js,android.img,android.boot.js","data":"android.settings.json,android.calendar.json,android.cards.json"},{"id":"widanclk","version":"0.03","files":"widanclk.info,widanclk.wid.js","type":"widget"},{"id":"backswipe","version":"0.04","files":"backswipe.info,backswipe.boot.js,backswipe.settings.js","data":"backswipe.json","type":"bootloader"},{"id":"torch","version":"0.11","files":"torch.info,torch.app.js,torch.img,torch.settings.js","data":"torch.settings.json"},{"id":"calculator","version":"0.07","files":"calculator.info,calculator.app.js,calculator.img"},{"id":"widbt_notify","version":"0.17","files":"widbt_notify.info,widbt_notify.wid.js,widbt_notify.settings.js","data":"widbt_notify.json","type":"widget"},{"id":"clock_info","version":"0.12","files":"clock_info.info,clock_info,clock_info.settings.js","data":"clock_info.json","type":"module"},{"id":"smpltmr","version":"0.09","files":"smpltmr.info,smpltmr.app.js,smpltmr.clkinfo.js,smpltmr.img"},{"id":"clkinfostopw","version":"0.03","files":"clkinfostopw.info,stopw.clkinfo.js","type":"clkinfo"},{"id":"runplus","version":"0.26","files":"runplus.info,runplus.app.js,runplus.img,runplus.settings.js,runplus_karvonen","data":"runplus.json"},{"id":"dtlaunch","version":"0.25","files":"dtlaunch.info,dtlaunch.app.js,dtlaunch.settings.js,dtlaunch.img","data":"dtlaunch.json","type":"launch"},{"id":"quicklaunch","version":"0.15","files":"quicklaunch.info,quicklaunch.settings.js,quicklaunch.boot.js,quicklaunch.app.js","data":"quicklaunch.json","type":"bootloader"},{"id":"kineticscroll","version":"0.04","files":"kineticscroll.info,kineticscroll.boot.js","type":"bootloader"},{"id":"widalarm","version":"0.02","files":"widalarm.info,widalarm.wid.js","type":"widget"},{"id":"alarm","version":"0.47","files":"alarm.info,alarm.app.js,alarm.img,alarm.settings.js","data":"alarm.settings.json"},{"id":"recorder","version":"0.43","files":"recorder.info,recorder.app.js,recorder.img,recorder.wid.js,recorder.clkinfo.js,recorder.settings.js","data":"recorder.json;recorder.log?.csv"},{"id":"bthrm","version":"0.19","files":"bthrm.info,bthrm.app.js,bthrm.recorder.js,bthrm.0.boot.js,bthrm.img,bthrm.settings.js,bthrm,bthrm.default.json","data":"bthrm.json"},{"id":"fastload","version":"0.06","files":"fastload.info,fastload.5.boot.js,fastload.settings.js","data":"fastload.json","type":"bootloader"},{"id":"agenda","version":"0.15","files":"agenda.info,agenda.app.js,agenda.settings.js,agenda.clkinfo.js,agenda.img","data":"agenda.settings.json"},{"id":"edgeclk","version":"0.03","files":"edgeclk.info,edgeclk.app.js,edgeclk.settings.js,edgeclk.img","data":"edgeclk.settings.json","type":"clock"},{"id":"activityreminder","version":"0.12","files":"activityreminder.info,activityreminder.app.js,activityreminder.boot.js,activityreminder.settings.js,activityreminder.alert.js,activityreminder,activityreminder.img","data":"activityreminder.s.json;activityreminder.data.json"},{"id":"twenties","version":"0.04","files":"twenties.info,twenties.boot.js","type":"bootloader"},{"id":"autoreset","version":"0.02","files":"autoreset.info,autoreset.boot.js,autoreset.settings.js","data":"autoreset.json","type":"bootloader"},{"id":"chargent","version":"0.07","files":"chargent.info,chargent.boot.js","data":"chargent.json","type":"bootloader"},{"id":"setting","version":"0.71","files":"setting.info,setting.app.js,setting.img","data":"setting.json"},{"id":"boot","version":"0.61","files":"boot.info,.boot0,.bootcde,bootupdate.js","type":"bootloader"},{"id":"mysetup","version":"0.02","files":"mysetup.info,backswipe.json,autoreset.json,dtlaunch.json,fastload.json,quicklaunch.json,messages.settings.json,widbt_notify.json,recorder.json,setting.json","type":"settings"},"BANGLEJS2","2v22.8",670000,1717449936,70418672]
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v22.8 (c) 2024 G.Williams
>OK
OK
Compacting Bank 1...|
Compacting Bank 2.../
Uncaught Error: Can't read property 'includes' of undefined
 at line 5 col 292 in .bootcde
...);s.clockHasWidgets=_clkApp.includes("Bangle.loadWidgets");r...
                              ^
>

```

[bootcde (1).txt](https://github.com/user-attachments/files/15540083/bootcde.1.txt) (Decoded)
[bootcde.txt](https://github.com/user-attachments/files/15540084/bootcde.txt)
[_``d`0.txt](https://github.com/user-attachments/files/15540085/_.d.0.txt) (corrupted .boot0 I believe)

gfwilliams, 2024-06-04:
Ok, thanks for the report. So it looks like compaction breaks it?

I don't have time to look into this at the moment so I've just removed the `iflash` automatic build from GitHub so it doesn't confuse normal users.

gfwilliams, 2024-06-04:
The last 'cutting edge' build for this is still available at https://www.espruino.com/binaries/travis/9898ae3a57448e441e912b56ba6b28d3ff8e739c/

thyttan, 2024-06-04:
Don't know if this is related. But on the regular fw 2v22.15 available from the app loader fw updater I get the error below:

```
OK
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v22.15 (c) 2024 G.Williams
>OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v22.15 (c) 2024 G.Williams
Execution Interrupted
at line 112 col 21 in bootupdate.js
let getPriority = /.*\.(\d+)\.boot\.js$/;
                  ^
in function called from line 123 col 2 in bootupdate.js
});
 ^
>OK
OK
OK
OK
OK
OK
OK
OK
OK
OK
>
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v22.15 (c) 2024 G.Williams
>
Storage Updated!
```

I didn't notice myself doing anything to cause the interruption, but it's not out of the question.

thyttan, 2024-08-15:
Additional report by @devsnd on the espruino forum: https://forum.espruino.com/conversations/397606/?offset=25#17485097

closed 2025-05-09 by gfwilliams

gfwilliams, 2025-05-09:
I was just following up on this, and I could easily reproduce by doing "Install Default Apps" in the app loader and then creating a file that can be cleaned and compacting:

```
s=require("Storage");
s.write("test.js","0");
s.write("test.js","1");
s.write("test2.js","0");
s.write("test2.js","1");
s.compact()
```

But if I do this on a special build on the PC which mirrors the addresses in Storage and everything in it, everything worked perfectly (despite all the log messages being identical).

After a lot of digging, it was actually a bug in `jshFlashWrite` when writing over 2kbytes - which means it affects all nRF52-series devices! I've just put a fix in, and I'll update https://forum.espruino.com/conversations/397606/?offset=25#17485097 with a new firmware

commit 840271c1 (2025-05-09): nRF52: Fix unreliable internal flash writes when writing >2k (big issue for compaction) (fix #2509)