espruino/Espruino #2524: nrf5x_dfu/main.c highlight

d3nd3, 2024-06-27:
https://github.com/espruino/Espruino/blob/a0740a24cc0fdb6fa731c544b722564368f5712c/targets/nrf5x_dfu/main.c#L373
My calculation is that every line which uses : 
`&& (r&0xF)==0`
Will never be reached, because of https://github.com/espruino/Espruino/blob/a0740a24cc0fdb6fa731c544b722564368f5712c/targets/nrf5x_dfu/main.c#L367

I hope I am wrong though, just I can't erase it from my mind.

gfwilliams, 2024-06-27:
To get past 367, `r&0xF` is either 0 or 4?

So that if statement can be reached just fine?

Definitely when you first apply power to a flat watch it doesn't turn on, which implies to me that line is working??

closed 2024-06-27 by gfwilliams

d3nd3, 2024-06-27:
Alrightey, I think I was mixing up the % operator, You're right there.