espruino/Espruino #2532: Stuck uploading firmware v23

dror3go, 2024-07-11:
I've tried to upgrade the firmware to the latest v23, and the progress bar gets stuck at about 80%.
I've waited for a few minutes every time, and it doesn't finish.
I then long-pressed the button on my Bangle.js2 to return to the home screen.
Repeated the process several times, same results. I once tried it after uninstalling fontall extension, which I've suspected is eating the storage on the watch.

Notifications from the phone stopped vibrating on the watch.

Any hints on what can I try?

gfwilliams, 2024-07-12:
It could be storage, but it actually allocates the file right at the start, so usually if there isn't enough space it'd fail at the beginning.

Where are you uploading from? Is it the App Loader inside Gadgetbridge?

If it's Gadgetbridge the communications from the App Loader can be a bit unreliable on some phones, so you could try disconnecting Gadgetbridge and then going to https://banglejs.com/apps/ directly and updating firmware from there?

If that fails there's always the DFU mode at https://www.espruino.com/Firmware+Update#nrf52 but if you can get the app loader working that's better!

dror3go, 2024-07-12:
Upgrading from a computer's browser worked - thanks a lot!

closed 2024-07-12 by dror3go