espruino/Espruino #2561: Compiling C code to run on ESP32

montanag, 2024-10-01:
Also posted to the forum [here](https://forum.espruino.com/conversations/399862/)

# Issue

I'm using the `setWatch` function on my [esp32 board](https://www.amazon.com/dp/B087TNPQCV). How could one create a 'native' function to be called in the interrupt itself?

From the documentation
>   // Advanced: If the function supplied is a 'native' function (compiled or assembly)
>   // setting irq:true will call that function in the interrupt itself
>   irq : false(default)

`E.compiledC` is unavailable for the esp32 board (understandably so) and I can't seem to find very much information on how to accomplish this. I've also begun trying to create a library to make some native code available following the guide [here](https://github.com/espruino/Espruino/blob/master/libs/README.md), but have had issues compiling. 

# Context
I'm working with a [RobotDyn AC Dimmer](https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html) that I want to be controlled by my esp32 board running Espruino. I have setWatch() function watching the `Z-C` pin from the dimmer board. When the watch fires, I can issue a quick pulse to the `PSM` pin on the dimmer board to engage the triac. The triac will then remain engaged until the next `Z-C` firing. Configuring the delay between when the `Z-c` pin pulses and when I send my `PSM` pulse should allow me to control the brightness. The issue that I'm running into is that the pulse sent to the `PSM` pin is too slow, resulting in a poor signal to the dimming board and flickering of attached lights. Here is some example code:

```
// Define the required pins
const zcPin = D34;
const psmPin = D27;

// Configure the pins
pinMode(zcPin, 'input_pullup');
pinMode(psmPin, 'output');

setWatch(
        function() {
            // Send a brief 1ms pulse to the ZC pin
            digitalPulse(psmPin, true, 1);
        },
        zcPin,
        { repeat: true, edge:'rising', debounce: 0, hispeed : true}
)
```
## Oscilloscope readings

The yellow line is the reading from the `ZC` pin and the blue line is the reading from the `PSM` pin being output from the esp32. The yellow line is a constant and the blue line appears to be `running` and somewhat random. The blue line should be perfectly in sync with the yellow line and pulse immediately after the rising edge, but neither of these is happening. Additionally, the 1ms pulse being sent is also not 1ms.

[image] 
[image] 
[image]

## Postulated solution

My idea for a solution to this is to have a bit of native code compiled such that I can call if from within the interrupt itself and achieve the required performance to achieve a consistent dimming.

gfwilliams, 2024-10-02:
The actual compiler tool is at https://github.com/gfwilliams/EspruinoCompiler so you could run that locally and try and get it running with the ESP32's GCC? It just needs the flags to allow it to produce position independent code. The Web IDE has a setting for the compiler URL it uses.

One big issue is that on ESP32 I believe functions called from IRQs need to be in RAM. Obviously the one you upload would be in RAM, but anything it called (eg digitalPulse) wouldn't be so I'm not sure if this would be useful anyway.

Assuming you can run from RAM, you're not actually executing any JS, so you can just 'bind' digitalPulse which will be faster:

```
setWatch(
            digitalPulse.bind(null, psmPin, true, 1),  zcPin,
        { repeat: true, edge:'rising', debounce: 0, hispeed : true}
)
```

but you may still hit execution delays since you're waiting for the JS engine (and irq:true won't work).

Another option is the mains signal is actually very regular intervals. You can maybe just use `Pin.writeAtTime` but with the time set for the *next* crossing: https://www.espruino.com/Reference#l_Pin_writeAtTime

Meanwhile the timer for the previous crossing should hopefully be executing in the background.

I don't know what your mains frequency is, but let's assume the UK's 50Hz (=20ms), something like this might work:

```
setWatch(
        function(e) {
            psmPin.writeAtTime(1, e.time+0.02);
            psmPin.writeAtTime(0, e.time+0.021);
        },
        zcPin,
        { repeat: true, edge:'rising', debounce: 0, hispeed : true}
)
```

SimonGAndrews, 2024-10-02:
Hi @montanag ,  I have had problems with the digital pulse function on the ESP32 in the past and sadly gave up on it and manually made the pulses.  Sorry I haven’t tried it lately.  But if you have time and you can test just a simple digital pulse script with your scope that would be most usefull.  Cheers

montanag, 2024-10-02:
@SimonGAndrews Thank you for the response.

Unfortunately,sending the pulses manually is also too slow/inconsistent

Here is a modified version sending the pulses manually with the corresponding scope readings

```
// Define the required pins
const zcPin = D34;
const psmPin = D27;

// Configure the pins
pinMode(zcPin, 'input_pullup');
pinMode(psmPin, 'output');

setWatch(
        function() {
            // Send a brief pulse to the ZC pin
            D27.set(); D27.reset();
        },
        zcPin,
        { repeat: true, edge:'rising', debounce: 0, hispeed : true}
)
```

**Scope**: https://i.imgur.com/WWVKvY4.mp4

Notice that the blue line corresponding to the `PSW` pin is still out of sync as well, leading my to believe the esp32 is still reacting too slowly.

Additionally here is some more evidence to support this claim

Code:
```
while (1) {
    D27.set(); D27.reset();
}
```

**Scope:** https://i.imgur.com/QUiLkJc.mp4

Notice for this one that even when toggling the pin on/off as fast as possible with no other operations, we still only get about 5 or so cycles per `z-c` firing. This gives us very little room to work with here and would likely only leave native code execution in the interrupt left as a possible means to solve this. Additionally, the blue line here is still inconsistent, meaning we're not getting a perfect 50/50 square wave. I don't know why this is, but perhaps due to the gc or something else occasionally consuming some clock cycles.

@gfwilliams Thank you for your response as well. I will be reading through it and trying it now.

montanag, 2024-10-02:
@gfwilliams I read through your response, thank you for the suggestions. Also, I am in the US so mains frequency is 60Hz which means that zero-crossings are actually 120Hz=8.3333333333333ms. This frequency is also confirmed by the scope where I'm seeing roughly a 8.4ms delay between each `Z-C` pulse.

Trying to sync up the frequency with the mains, but instead writing to the pin ahead of the time seems like a good approach. I tried that with this

```js
// Define the required pins
const zcPin = D34;
const psmPin = D27;

// Configure the pins
pinMode(zcPin, 'input_pullup');
pinMode(psmPin, 'output');

setWatch(
    function(e) {
        psmPin.writeAtTime(1, e.time+0.016666666666667);
        psmPin.writeAtTime(0, e.time+0.017666666666667);
    },
    zcPin,
    { repeat: true, edge:'rising', debounce: 0, hispeed : true}
)
```

but am still seeing similar results to the other failed attempts on the scope. The blue line is still wandering and inconsistent. I've also tried a few more complicated solutions inspired by this idea, but none successful so far. Writing ahead like this still seems to take up too many operations. I've tried writing for just the very next cycle, and also things like removing the watch and writing for the next 100 or so cycles. None of these attempts worked and all produce wandering/random scope readings. I'm really thinking the only way here would be to get code called directly from within the interrupt itself. If there was a very precise to use `analogWrite` to start at a very specific time that could possible work, but the mains frequency and the PWM signal would still wander over time and require corrections. Due to the way the triac works, the `PWM` frequency needs to be very precisely in sync with the mains frequency.

I'll poke around a bit now with trying the compiled code approach, but as you stated, this might be a dead end for other reasons as well.

gfwilliams, 2024-10-03:
Ok, that's a shame. You can make things faster by pre-binding all the known values:

```
setWatch(
    function(a,b,e) {"ram"
        a(e.time+0.01667);
        b(e.time+0.01767);
    }.bind(null,psmPin.writeAtTime.bind(psmPin,1),psmPin.writeAtTime.bind(psmPin,0)),
```

If you're uploading to flash, function code gets stored and run from flash so adding `"ram"` can help there too.

(Not tested, but should work) but it feels like what you're hitting here is not so much execution speed (although it's slow) so much as a failure of the interrupts to run on time because presumably other stuff like Wifi (or RTOS tasks) might be happening at the same time.

As a test you could compile your own firmware with a `jshPinOutput(27,1);jshPinOutput(27,0);` right in the IRQ handler `gpio_intr_handler`: https://github.com/espruino/Espruino/blob/master/targets/esp32/jshardware.c#L115C16-L115C33
GitHub Actions will do the build automatically if you just add the code in a forked repo.

Note that you could also write C/assembly code to poke the registers directly, build it into a binary, and then include that binary in your JS with `E.nativeCall` - that's an option that doesn't require you to do a custom build or get the compiler working. 

But the issue could be the timer itself (used for writeAtTime/pwm/digitalPulse) which is what @SimonGAndrews  is mentioning was his problem. This uses interrupts and a hardware timer:

https://github.com/espruino/Espruino/blob/master/targets/esp32/rtosutil.c#L186
https://github.com/espruino/Espruino/blob/master/targets/esp32/jshardwareTimer.c#L32-L42

If you do `analogWrite(psmPin,0.1,{freq:60,soft:true})` this should use that software timer, and if that's not reliable but the build with `jshPinOutput` is then I think if it could be fixed (maybe just a matter of changing priority levels?) the `writeAtTime` code should work well for you

montanag, 2024-10-03:
Thank you for the continued replies.

I attempted the pre-binding approach you suggested, which got somewhat strange results. The blue line still wanders and is inconsistent.

**Code:**
```js
// Define the required pins
const zcPin = D34;
const psmPin = D27;

// Configure the pins
pinMode(zcPin, 'input_pullup');
pinMode(psmPin, 'output');

setWatch(
    function(a,b,e) {"ram"
        a(e.time+0.01667);
        b(e.time+0.01767);
    }.bind(null,psmPin.writeAtTime.bind(psmPin,1),psmPin.writeAtTime.bind(psmPin,0)),
    zcPin,
    { repeat: true, edge:'rising', debounce: 0, hispeed : true}
)
```

**Scope:** https://i.imgur.com/JxvWHZX.mp4

> (Not tested, but should work) but it feels like what you're hitting here is not so much execution speed (although it's slow) so much as a failure of the interrupts to run on time because presumably other stuff like Wifi (or RTOS tasks) might be happening at the same time.

I think you might be right here, especially so sense I'm not getting consistent output either. The blue line is "running" on the scope, meaning sometimes it is written sooner or later than others, presumably due to the other ongoing processes you mentioned.

> As a test you could compile your own firmware with a jshPinOutput(27,1);jshPinOutput(27,0); right in the IRQ handler gpio_intr_handler: https://github.com/espruino/Espruino/blob/master/targets/esp32/jshardware.c#L115C16-L115C33
GitHub Actions will do the build automatically if you just add the code in a forked repo.

I've forked the repository [here](https://github.com/montanag/Espruino) and modified the `gpio_intr_handler` function [here](https://github.com/montanag/Espruino/blob/master/targets/esp32/jshardware.c#L115-L118).

The actions were disabled be default after the fork, but I've re-enabled and ran a workflow after my commit. That run exists [here](https://github.com/montanag/Espruino/actions/runs/11170743800) which I then downloaded the [ESP32 Artifact](https://github.com/montanag/Espruino/actions/runs/11170743800/artifacts/2013376274) from.

I then re-flashed my esp32 with this new firmware. I used the old bootloader and old partitions (not sure what these do exactly here).

```sh
python3 $ESPTOOL_DIR/bin/esptool.py                     \
        --chip esp32                                    \
        --port /dev/ttyUSB0                             \
        --baud 921600                                   \
        --after hard_reset write_flash                  \
        -z                                              \
        --flash_mode dio                                \
        --flash_freq 40m                                \
        --flash_size detect                             \
        0x1000 $ESPURINO_DIR/bootloader.bin             \
        0x8000 $ESPURINO_DIR/partitions_espruino.bin    \
        0x10000 $ESP_CUSTOM_BIN # new firmware
```

After that, I could run this small piece of code.

**Code:**
```js
// Define the required pins
const zcPin = D34;
const psmPin = D27;

// Configure the pins
pinMode(zcPin, 'input_pullup');
pinMode(psmPin, 'output');

setWatch(
    function() {
        // Do nothing - the firmware should now activate the pin directly
    },
    zcPin,
    { repeat: true, edge:'rising', debounce: 0, hispeed : true}
)
```

And this actually seems to be working, or at least shows much promise. I am seeing _extremely_ brief pulses on both the rising and falling edge of the `Z-C` pulse. These brief pulses are also consistent in their timing. In fact, these pulses are so extremely brief that they don't even engage the triac save for once every few seconds for a quick flash. They're about 10 micro seconds and 1 microsecond on the rising and falling edge respectively to be exact.

**Whole `Z-C` pulse:**
[image]

**Zoomed in rising edge:**
[image]

**Zoomed in falling edge:**
[image]

What would be the next steps here? Could I write a digitalPulse function in native code that could be called from the js code then? These next steps here are where I get a little lost, and couldn't have even gotten this far without the help so far.

gfwilliams, 2024-10-04:
That's great! Well, the nice thing is it at least means that the `e.time` values are going to be correct because the IRQ is working right.

The problem is you can't really get away with just waiting for 1/100th sec in the IRQ because it'll almost certainly screw up wifi. So you'd need to set a hardware timer.

As far as I know the hardware timers are https://docs.espressif.com/projects/esp-idf/en/v4.3/esp32/api-reference/peripherals/timer.html, one of which is already used for writeAtTime/digitalPulse/etc: https://github.com/espruino/Espruino/blob/master/targets/esp32/jshardwareTimer.c#L32-L42

So you could use TIMER_GROUP_1 I think, but whatever is stopping the built-in timers from working may well stop even a compiled version from working right too.

Ok, did you get a chance to try the:

```
analogWrite(psmPin,0.1,{freq:60,soft:true})
```

command to see if it's regular? Obviously it won't align with the mains though.

I guess ideally we can fix the existing timers, and then the `writeAtTime` solution should work reliably...

gfwilliams, 2024-10-04:
Do you have other code running on the ESP32? I just rigged this up on an ESP32 with:

```
setWatch(function() {                                                          
  D27.set();D27.reset();                                                       
}, D25, {repeat:1,edge:0})  
```

And I see something more reliable than you were getting it seems: https://photos.app.goo.gl/FSySp63radyMfC7M6

However the second I try and use timers it all goes wrong, as you found. However I spotted a bug in the ESP32 Espruino code - it never actually stopped the timer from running, so subsequent calls to digitalPulse/writeAtTime/etc all executed the first call almost immediately rather than at the correct time.

If you use the latest cutting edge build (or build your own) I believe it's fixed now. This works well for me:

```
zcPin=D25;
psmPin=D27;
pinMode(zcPin, 'input_pullup');
pinMode(psmPin, 'output');
AMT=0.003;
setWatch(
    function(w,e) { "ram" // ensure it's in RAM and pretokenised
        w(1, e.time+0.016); // turn off before zero crossing
        w(0, e.time+0.0167+AMT); // turn on after
    }.bind(null,psmPin.writeAtTime.bind(psmPin)),
    zcPin,
    { repeat: true, edge:'rising', debounce: 0}
)
```

https://photos.app.goo.gl/rz2PFHdtXA49ivr16

There's still the odd glitch, but that could just be a bad connection from my signal source

montanag, 2024-10-04:
## Analog write

I tried the analog write a while back and that always actually worked for me just fine, it's just out of sync with the mains as you mentioned. Trying again today is seems to wander more than it did before for whatever reason? It would always "run" from the mains frequency but seems to be doing it faster now for whatever reason. I omitted the `soft: true` from the `analogWrite` but adding it makes the signal "drift/run" faster than it already is. This might be due to the mains frequency not being exactly 60Hz either. My scoped is measuring anywhere from 119.8-120Hz for the zero crossings.

Code:
```js
// Define the required pins
const zcPin = D34;
const psmPin = D27;

// Configure the pins
pinMode(zcPin, 'input_pullup');
pinMode(psmPin, 'output');

setWatch(
    function() {
        // Do nothing
    },
    zcPin,
    { repeat: true, edge:'rising', debounce: 0, hispeed : true}
)

analogWrite(psmPin,0.1,{freq:120})
```

**Scope:** https://i.imgur.com/1Uvi5E9.mp4

## Other code running

> Do you have other code running on the ESP32? ...And I see something more reliable than you were getting it seems

Not that I'm aware of, I re-flashed (using the nightly build from this [commit](https://github.com/espruino/Espruino/actions/runs/11176658658)) and ran `espruino --board esp32 -b 115200 --port /dev/ttyUSB0 ac-dimmer-control.js` immediately after and see [this](https://i.imgur.com/kyeGyzb.mp4).

> If you use the latest cutting edge build (or build your own) I believe it's fixed now. This works well for me:

I reflashed with your exact code (changing the pins I was using to match) and still unfortunately saw poor output. I ran the file with `espruino --board esp32 -b 115200 --port /dev/ttyUSB0 myFile.js` and see [this](https://i.imgur.com/wCaWT5v.mp4). This was using the firmware from [this build](https://github.com/espruino/Espruino/actions/runs/11176658658) with new firware, partions, and bootloader. I'm really not sure why I'd be seeing different output than you here, I'll poke around some more.

montanag, 2024-10-04:
Here are some test I conducted with the timers. From what I understand, there are only a limited number of timers that can be set, so these tests might not really provide anything useful if these go over that limit. All the below snipets were ran individually.

```js
// Works as expected
const time = getTime();
const interval = 1;
const delay = 0;
for (i=1; i<10; i++) {
    psmPin.writeAtTime(1, delay + time+i*interval);
    psmPin.writeAtTime(0, delay + time+i*interval+interval*0.5);
}

// Works for the first ~8 flashes
const time = getTime();
const interval = 1;
const delay = 0;
for (i=1; i<100; i++) {
    psmPin.writeAtTime(1, delay + time+i*interval);
    psmPin.writeAtTime(0, delay + time+i*interval+interval*0.5);
}

// Works  as expected
const time = getTime();
const interval = 1;
const delay = 10;
for (i=1; i<10; i++) {
    psmPin.writeAtTime(1, delay + time+i*interval);
    psmPin.writeAtTime(0, delay + time+i*interval+interval*0.5);
}

// Produces a weird inconsistent square wave - behaves unexpectedly
const time = getTime();
const interval = 0.1;
const delay = 0;
for (i=1; i<100; i++) {
    psmPin.writeAtTime(1, delay + time+i*interval);
    psmPin.writeAtTime(0, delay + time+i*interval+interval*0.5);
}
```

montanag, 2024-10-04:
I realized that the code sample above is for 60Hz zero-crossings, where I have 120Hz zero crossings.

I tried this piece again:
```js
zcPin=D25;
psmPin=D27;
pinMode(zcPin, 'input_pullup');
pinMode(psmPin, 'output');
AMT=0.003;
setWatch(
    function(w,e) { "ram" // ensure it's in RAM and pretokenised
        w(1, e.time+0.008333333333); // turn off before zero crossing
        w(0, e.time+0.008333333333+AMT); // turn on after
    }.bind(null,psmPin.writeAtTime.bind(psmPin)),
    zcPin,
    { repeat: true, edge:'rising', debounce: 0}
)
```
And am seeing much better results. Still not perfect, but much better than I've been able to see so far. This is running the new firmware.

**Scope:**  https://i.imgur.com/1qBPVOn.mp4

You can see that I'm getting a much more consistent square wave here, though with the occasional anomaly. With these settings the light flashes kinda weirdly but mostly remains off. I'll try some adjustments here to see what I can get.

montanag, 2024-10-04:
I wrote a small piece of code here to play around and see if I could actually control the brightness.

```js
zcPin=D25;
psmPin=D27;
pinMode(zcPin, 'input_pullup');
pinMode(psmPin, 'output');
const AMT = 0.0001;
const period = 0.008333333333333
const offset = period * 0.9;
const setTime = offset + period;
const resetTime = setTime + AMT;
setWatch(
    function(w,e) { "ram" // ensure it's in RAM and pretokenised
        w(1, e.time+setTime); // turn off before zero crossing
        w(0, offset+resetTime); // turn on after
    }.bind(null,psmPin.writeAtTime.bind(psmPin)),
    zcPin,
    { repeat: true, edge:'rising', debounce: 0}
)
```

**Scope:** https://i.imgur.com/bYcCjpm.mp4

The `offset` variable is effectively what I would use to control when the triac engages and hence the brightness. From looking at the scope readings, I feel as though the signal I'm getting is still to jittery to effectively be useful. I'm not sure why it's jittering like this, however these are the best results I've seen yet. This also was with the firmware mentioned in comments above.

gfwilliams, 2024-10-07:
I guess I was running off a 60hz square wave on mine, not 120.

That higher frequency output in soft PWM might be to some 'bodge factor' that was left in the ESP32 code - I wonder whether that should actually be zero, or at least smaller: https://github.com/espruino/Espruino/blob/master/targets/esp32/rtosutil.c#L177

> From what I understand, there are only a limited number of timers that can be set

We have ~16 timers in a queue I think, so when you add more it just waits for one to complete before adding the next.

However your code in the last example looks a bit off?

```
setWatch(
    function(w,e) { "ram" // ensure it's in RAM and pretokenised
        w(1, e.time+setTime); // turn off before zero crossing
        w(0, offset+resetTime); // turn on after
    }.bind(null,psmPin.writeAtTime.bind(psmPin)),
    zcPin,
    { repeat: true, edge:'rising', debounce: 0}
)
```

So you turn it on at `e.time+setTime` which looks good, but then ` w(0, offset+resetTime)` is not related to the current time at all. I think maybe you meant `e.time+resetTime`?

But if that doesn't work I'm not really sure what to suggest. You could do your own firmware build with the jshPinOutput and then `jstPinOutputAtTime` after it to turn it off at the correct time, and maybe if the timer is working ok now, that would be enough?

Just a note on the other code you did - Espruino parses as it runs, so if you do something like `0.008333333333` in the function that gets called very often it's having to interpret each character of that number each time. Realistically even if you did `0.0083` it's probably not going to make much difference to the value but would execute faster. The last bit of code you posted wouldn't have that problem though as it's in a variable.

montanag, 2024-10-07:
Thanks again for the continued replies.

I corrected the code from the error you pointed out:
```js
const zcPin=D25;
const psmPin=D27;
pinMode(zcPin, 'input_pullup');
pinMode(psmPin, 'output');
const AMT = 0.0001;
const period = 0.008333333333333
const offset = period * 0.9;
const setTime = offset + period;
const resetTime = setTime + AMT;
setWatch(
    function(w,e) { "ram" // ensure it's in RAM and pretokenised
        w(1, e.time+setTime); // turn off before zero crossing
        w(0, e.time+resetTime); // turn on after
    }.bind(null,psmPin.writeAtTime.bind(psmPin)),
    zcPin,
    { repeat: true, edge:'rising', debounce: 0}
)
```

Unfortunately, I still am not really seeing usable results.

For the time being, I am going to table this, but I'll probably revisit later and try to write some kind of binary or something that can be called from espruino and use some kind of shared variable to pass in the desired parameters. I'm not sure how to do that just yet but will worry about it later.

For now, I've got it working in c and just for the curious have the following code for that. It's kind of a plug of sorts for espruino too with just how much more involved the same c code is in comparison. This code is also still pretty messy right now.

```c
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include <time.h>
#include <sys/time.h>
#include "esp_timer.h"
#include <driver/gptimer_types.h>
#include <driver/gptimer.h>

#define ZC_GPIO 25  // Set ZC pin
#define PSM_GPIO 27  // Set PSM pin

gptimer_handle_t gptimer = NULL;

static void busyPulse(gpio_num_t gpio_num, uint32_t us_delay) {
    gpio_set_level(gpio_num, 1);
    esp_rom_delay_us(us_delay);
    gpio_set_level(gpio_num, 0);
}

static bool alarm_handler(gptimer_handle_t timer, const gptimer_alarm_event_data_t *edata, void *user_data) {
    // Stop and reset the timer
    gptimer_stop(gptimer);
    gptimer_set_raw_count(gptimer, 0);

    // Busy pulse the PSM pin for 50us
    busyPulse(PSM_GPIO, 50);

    return true;
}

int lastTime = 0;
int debounce = 1000;
static void isr(void* arg)
{
    struct timeval tv;
    gettimeofday(&tv, NULL);
    int time = tv.tv_sec * 1000 * 1000 + tv.tv_usec;
    if (time < lastTime + debounce) {
        lastTime = time;
        return;
    }

    lastTime = time;

    // Start the timer
    ESP_ERROR_CHECK(gptimer_start(gptimer));
}

void app_main(void)
{
    // Set the pin directions
    gpio_set_direction(PSM_GPIO, GPIO_MODE_OUTPUT);
    gpio_set_direction(ZC_GPIO, GPIO_MODE_INPUT);
    gpio_set_pull_mode(ZC_GPIO, GPIO_PULLUP_ONLY);

    //// Timer stuff

    // Create the configuration for a new 1us timer
    printf("Creating timer configuration\n");
    gptimer_config_t timer_config = {
        .clk_src = GPTIMER_CLK_SRC_DEFAULT,
        .direction = GPTIMER_COUNT_UP,
        .resolution_hz = 1 * 1000 * 1000, // 1MHz, 1 tick = 1us
    };

    // Create the new timer
    printf("Creating timer\n");
    ESP_ERROR_CHECK(gptimer_new_timer(&timer_config, &gptimer));

    // Create the callback information
    printf("Creating timer callback information\n");
    gptimer_event_callbacks_t callback = {
        .on_alarm = alarm_handler
    };

    // Register the callback
    printf("Registering timer callback\n");
    ESP_ERROR_CHECK(gptimer_register_event_callbacks(gptimer, &callback, NULL));

    // Enable the timer
    printf("Enabling timer\n");
    ESP_ERROR_CHECK(gptimer_enable(gptimer));

    // Create the configuration for a timer alarm
    printf("Creating alarm config\n");
    gptimer_alarm_config_t alarm_config = {
        .alarm_count = 4000, // Timer count value for which an alram event will be triggered
    };

    // Set the alarm
    printf("Setting alarm\n");
    ESP_ERROR_CHECK(gptimer_set_alarm_action(gptimer, &alarm_config));

    //// Pin ISR stuff

    // Set the interrupt type
    ESP_ERROR_CHECK(gpio_set_intr_type(ZC_GPIO, GPIO_INTR_POSEDGE));

    ESP_ERROR_CHECK(gpio_install_isr_service(ESP_INTR_FLAG_EDGE));

    // Register the ISR
    ESP_ERROR_CHECK(gpio_isr_handler_add(ZC_GPIO, isr, NULL));

    int startVal = 4000;
    int val = startVal;
    while(1) {
        val += 10;
        if (val > 8000) {
            val = startVal;
        }
        gptimer_alarm_config_t alarm_config = {
            .alarm_count = val, // alarm in next 1s
        };
        gptimer_set_alarm_action(gptimer, &alarm_config);

        printf("Hello World %d\n", val);

        vTaskDelay(10 / portTICK_PERIOD_MS);
    }
}
```

montanag, 2024-10-08:
## Using E.nativeCall

Ok, so after some more digging, I've at least started to answer some of my own questions. This may be verbose, but some of that is for my own benefit so I or others can come back later. I realize some of the questions here are not necessarily related to espruino and more related to general c questions. I'm not a very experienced c programmer.

### Write a c program

A simple program to add two numbers.

**Note:** It would be great if I could omit the `app_main` here and compile this as a library or something.

```c
static double add_two_numbers(double first, double second) {
    return first + second;
}

double app_main() {
    // Function must be used in some way to make it's way into the binary and not be optimized out
    return add_two_numbers(3.3,4.4);
}
```

### Compile the c program

I'm using `idf.py` from the command line, so to compile the program I've just ran `idf.py build`

### Get the function offset from the elf

Next, I get the function offset from the built `.elf` file. To do so I run the following:

```
~/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/xtensa-esp32-elf-objdump -D build/add-two-numbers.elf | grep add_two_numbers
400d6214 <add_two_numbers>:
400d623b:       fffda5          call8   400d6214 <add_two_numbers>
```

Which gives a few numbers, I'm not sure which one I need to use here, but let's say it's `400d6214`

### Reflash the ESP32

After this, I can reflash the chip and also upload the binary to a particular location on the flash.

```bash
python3 $ESPTOOL_DIR/bin/esptool.py                     \
        --chip esp32                                    \
        --port "/dev/ttyUSB0"                           \
        --baud 921600                                   \
        write_flash                                     \
        -z                                              \
        --flash_mode "dio"                              \
        --flash_freq "40m"                              \
        --flash_size "8MB"                              \
        0x1000 $ESPURINO_DIR/bootloader.bin             \
        0x8000 $ESPURINO_DIR/partitions_espruino.bin    \
        0x10000 $ESPURINO_DIR/espruino_esp32.bin        \
        0x200000 /home/montana/esp/tests/add-two-numbers/build/add-two-numbers.bin
```

I'm not sure if these addresses are good to use or not. I assumed the first 3 are special and wanted to keep them exactly where they're positioned. I chose `0x200000` because it seemed to not overlap.

### Writing an espruino program

Next, I wrote a simple program to try and call the native code.

```js
console.log("Hello")

const res = E.nativeCall(0x402D6214,'double (double,double)')(1.1, 2.2);
console.log(`Result: ${res}`);
```

Then I simply called this program from the command line with `espruino --board esp32 -b 115200 --port /dev/ttyUSB0 call-c-code.js`. The output was

```
--] Hello
```

So  it must have crashed from the native call.

gfwilliams, 2024-10-08:
Hi - I think realistically by pulling the whole program code in you're making like difficult for yourself - 175K is way too much so there's probably loads of initialisation code in there - and if it's not run the code you're running may break, and if it is run, it'll probably break Espruino.

You also can't run the code from flash if you're planning on using it from an IRQ.

Basically in the C code you could change the GPIOs by changing stuff at the relevant addresses by looking at the ESP32 reference, and could configure timers the same way - but you're not going to be able to call into the RTOS for anything, and trying to set up an IRQ to call into the code stored in RAM could be really flakey.

IMO I'd skip idf.py (as you can't really use the built-in functions anyway if they call into the RTOS) - and try using the compiler directly. Info on doing that for ARM at https://www.espruino.com/Assembler#compiling-c-code and hopefully you can work out how to use the xtensa one in a similar way.

MaBecker, 2024-10-08:
check this page too.

http://cholla.mmto.org/esp32/bootrom/lightning.html

montanag, 2024-10-10:
> Hi - I think realistically by pulling the whole program code in you're making like difficult for yourself - 175K is way too much so there's probably loads of initialisation code in there - and if it's not run the code you're running may break, and if it is run, it'll probably break Espruino.

That makes sense. I think I'm going to table this for right now and come back later when I have a little more experience with c code and the whole ecosystem here. Thanks for all the responses and working on this with me!