espruino/Espruino #2573: nrf52840 lost USB-port

louisvangeldrop, 2024-10-21:
I have installed espruino_2v19.119_nrf52840.uf2 on a nice!nano board. It works nicely, however there is n o USB-connection anymore with my Windows laptop. Also resetting the device to the bootloader-mode using the RST to GND link is not workimng anymore.

Flashing the device has become impossible.

louisvangeldrop, 2024-10-21:
I have found "espruino_2v19.119_nrf52840.hex" somewhere at the internet and converted into uf2 file:
uf2conv.py espruino_2v19.119_nrf52840.hex --family 0xADA52840 --convert --output espruino_2v19.119_nrf52840.uf2

Then I copied the uf2-file to the board using the UF2 folder. 
Through the webbluetooth interface I can now use the beautiful espruino interpreter.

gfwilliams, 2024-10-21:
ok, well I think that's not a firmware issue then!

You just took a firmware that wasn't designed to exist alongside a bootloader and flashed it to a device with a bootloader - so it's been overwritten now.

But at least you have Espruino! You might be able to use Espruino's flash write commands to reinstate the bootloader (at the end of the address space) but that's not really something I can help with.

louisvangeldrop, 2024-10-26:
I have built a espruino 2v24.120 version for the promicro board using: https://github.com/jeffmer/EspruinoProjects/blob/main/boards/PROMICRO.py

It results in 3 files: espruino_2v24.120_promicro.app_hex, espruino_2v24.120_promicro.elf and espruino_2v24.120_promicro.hex.
In order to prevent a bootloader problem again, which file should I copy to the UF2-disk?

gfwilliams, 2024-10-28:
You can copy in the app_hex but you'll have to ensure the softdevice matches. The hex contains everything (softdevice+app+config) I believe but I guess the config might overwrite something for the bootloader

louisvangeldrop, 2024-10-28:
I have copied then app_hex file to the UF2-drive. However the espruino app is not installed at the NRF52840

louisvangeldrop, 2024-11-19:
The board I am using is a Nice!Nano V2.0 compatible with Promicro.
After successfully building the uf2file using:
uf2conv.py espruino_2v24.120_promicro.app_hex --family 0xADA52840 --convert --output espruino_2v24.120_promicro.app.uf2

I copied the generated uf2-file to the board. Now I can connect to the board either using web-bluetooth or webserial. However the response is: Connected to PROMICRO 5df1 (No response from board). And the communication is stuck.