espruino/Espruino #2580: require('hmac') is flaky (Bangle.js 2)

rogueeve575, 2024-12-01:
Yes AFAICT, it's specifically just that library.

### Test case:

First I uploaded this simple module, TestLibrary.js:
```
const hmac = require('hmac');

function foo() {
   var hInstance = hmac.SHA256(E.toArrayBuffer("passcode"));
   console.log("Created HMAC instance: " + hInstance);
}

exports.foo = foo; 
```

Now using the Web IDE, I run this script, just uploading it to RAM:
 
```
//require('hmac');
var test = require('TestLibrary.js');
test.foo();

```

Now we get an error from TestLibrary.js; `Module "hmac" not found`. Note that essentially the same code works fine when ran directly.

But if we un-comment the "useless" `require('hmac')` in the RAM script so that "hmac" module is already loaded though, suddenly no more problems.

And next, once we finish our RAM script and package it up into an app and run from the Launcher on the watch like normal, the workaround no longer workarounds -- now we get a "not found" crash on the _first_ require('hmac'); like the Bangle can't find hmac at all.

This doesn't seem to happen if I do the same thing with any other library, such as "crypto".

Lastly, I should perhaps mention that because I have some "bigger" app designs in WIP, too big to develop entirely in the Web IDE, and couldn't get the espruino cli tool to detect BLE due to the whole Noble issue (EspruinoTools repo #63), I've rolled my own quick-and-dirty installer for my app testing which is just a script that generates js code to be pasted into the REPL. For completeness's sake, the REPL paste that was used to install "TestLibrary.js" output from this tool looks like:

```
{var install = function() { var r; console.log('\n\n');
function inst0() {
return require("Storage").write("TestLibrary.js", atob('CmNvbnN0IGhtYWMgPSByZXF1aXJlKCdobWFjJyk7CgpmdW5jdGlvbiBmb28oKSB7CiAgIHZhciBoSW5zdGFuY2UgPSBobWFjLlNIQTI1NihFLnRvQXJyYXlCdWZmZXIoInBhc3Njb2RlIikpOwogICBjb25zb2xlLmxvZygiQ3JlYXRlZCBITUFDIGluc3RhbmNlOiAiICsgaEluc3RhbmNlKTsKfQoKZXhwb3J0cy5mb28gPSBmb287IAo='));
}
r = inst0();

if (!r) { console.log('failed to install TestLibrary.js'); return; }
console.log('install TestLibrary.js: ' + r);
console.log("Result: " + r);return r;
} install(); }

```

For a full app install, there are additional files written for the app icon, info file etc.

I haven't had any other problems from using this tool, so I don't _think_ it's the cause of this issue, but since most apps likely get the most testing being installed through the App Loader, installing stuff this way might potentially have exposed lesser-seen behavior somehow.

However, I believe that the libraries such as "hmac" are special libraries in that they're native code provided by the runtime and don't have any sort of corresponding file on the watch, right? Which actually confuses me even more how such a library could go missing.

rogueeve575, 2024-12-02:
So silly me, I just noticed on this page:
https://www.espruino.com/hmac

That the "hmac" module **_is_** actually a JavaScript library. When I download it and install it on to the watch with my little installer script, it's accessible everywhere again.

By saving a script that uses HMAC (and was previously working) to a file and then viewing it again, I can confirm that the IDE was downloading this module itself and inlining it in to my code behind the scenes.

Although this process is seamless when using the IDE, I'd suggest modifying the above-mentioned HMAC documentation (and any other .js modules) a little to make it slightly more clear that the HMAC module is NOT a built-in one like the others. I only noticed by clicking the very small "hmac" link in the first sentence "This hmac (about modules) module" -- IMHO that link is just too small, making it a little confusing what's going on for people that may need to understand for whatever reason - it took a couple of days in my case.

Also, if this hadn't broken, I might not have realized that I was inlining "hmac" into every app that uses it, so mightn't have known to change it if I wanted it to be a proper shared lib.

gfwilliams, 2024-12-02:
To be fair, I believe every module page links to `About Modules` where it mentions the module name, eg:

https://www.espruino.com/hmac -> https://www.espruino.com/Modules

Which gives a pretty clear breakdown of how this works.

But I'm still a bit surprised - how did you upload TestLibrary? Because if you did it with the IDE or with the BangleApps app loader, it should inline the module.

So the issue is you made your own tool that just uploaded everything without automatically adding the extra module?

I think in future, if you want to get something to upload, you can try using the CLI with `-o out.js` which I think should provide you with the actual JS you need to upload. However obviously getting the CLI to actually work directly would be preferable!

closed 2024-12-02 by gfwilliams