espruino/Espruino #2585: ESP32 WROOM32 internal temperature

0wwafa, 2024-12-17:
I have this device: 
[image]

I uploaded the latest firmware update 2v25.11

In a program I wrote some time ago I have E.getTemperature() to get the internal temperature.
But now I get it's NOT IMPLEMENTED?!

Why?
I don't know which firmware version I had before but the program was working!

```
>E.getTemperature()
ERROR: >> jshReadTemperature Not implemented
=NaN
> 
```

0wwafa, 2024-12-17:
In the source I read:
```
// the temperature from the internal temperature sensor
JsVarFloat jshReadTemperature() {
#if defined(STM32F1) || defined(STM32F4)
  // enable sensor
  ADC_TempSensorVrefintCmd(ENABLE);
  jshDelayMicroseconds(10);
  // read
  JsVarFloat varVolts = jshAnalogRead(JSH_ANALOG1 | JSH_ANALOG_CH17, false) / (JsVarFloat)65535;
  JsVarFloat valTemp = jshAnalogRead(JSH_ANALOG1 | JSH_ANALOG_CH16, false) / (JsVarFloat)65535;
  JsVarFloat vSense = valTemp * V_REFINT / varVolts;
  // disable sensor
  ADC_TempSensorVrefintCmd(DISABLE);
  return ((V_TEMP_25 - vSense) / V_TEMP_SLOPE) + 25;
#else
  return NAN;
#endif
}
```

But ESP32 WROOM has the sensor and in micropython it's working!

SimonGAndrews, 2024-12-18:
Looks like it was changed (file /targets/esp32/jshardware.c)  on April 26 this year in commit 973d26a. (see image of code change below). 
It seems there has always been a note saying the functionality used an undocumented library function.  But not clear why it was changed in April.

In 2018 [this](https://esp32.com/viewtopic.php?t=5875) was published on the subject.

The newer versions of the espressif IDF have examples of reading the sensor [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/temperature_sensor/temp_sensor).  But only for more recent processors eg c3 ,s3 etc.  In fact there is a comment in the example in IDF V4.3 
[image]
saying the ESP32 does not support the method.   But  we could try to  implement it in the Esp32C3 build if that would help.

Sorry not an answer for you @0wwafa , but a bit of info. 

[image]

gfwilliams, 2024-12-18:
Thanks @SimonGAndrews!

I'm not 100% sure why that was removed, but I believe this was merging in someone else's changes for IDF5 builds.

I've put that change back now so cutting edge builds should be ok - but it might be nice to get the proper sensor in for C3/etc

But I'm not quite sure what you were using getTemperature for but it varies wildly on the ESP32 so would need calibrating - mine reports 107 which I'm pretty sure isn't right.

closed 2024-12-18 by gfwilliams

0wwafa, 2025-01-15:
In the previous version it was working well. Now my program does now work :(

0wwafa, 2025-01-15:
P.S.
My esp32 says it's around 56 degrees and putting a finger on it "confirms it".
Why is it running so hot? With micropython it's quite cold.

gfwilliams, 2025-01-16:
Interesting - it could be BLE means it's not sleeping so much? I'm not sure.

ESP chips do run hot though - Wifi consumes a lot of power so if that's enabled it's just going to end up getting warm

commit 9cd8d5bd (2024-12-18): Revert change that removed ESP32 temperature sensor - fix #2585