espruino/Espruino #2588: Running mqtt on ESP32_ID4 cause "E (506885) task_wdt: Task watchdog got triggered."

MaBecker, 2024-12-29:
Find on ESP32_IDF4 build, It does not happen for ESP32 builds. Every 60sec when the mqtt module checks the connection it happens.  Can anyone else please check this or try this for a ESPC3_IDF4 or ESP32S6_IDF4 board.

Striped down sample coding

```js
var wifi=require("Wifi");
var ssid="your ssid";
var password="your ssid password";

wifi.disconnect(() => {
  wifi.connect(ssid, {password: password}, function(e) {
    if (e === null ) {
      print("wifi: connected");
      print("mqtt: connect to broker");
      mqtt = require("MQTT").connect({host: "test.mosquitto.org"});
      mqtt.on('connected', function() {
        print("mqtt: connected");
      });
    } else {
      print("wifi: connect error:",e);
    }
  });
});

```

output

```c
>;31mE (436245) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:m
;31mE (436245) task_wdt:  - IDLE0 (CPU 0)m
;31mE (436245) task_wdt: Tasks currently running:m
;31mE (436245) task_wdt: CPU 0: espruinoTaskm
;31mE (436245) task_wdt: CPU 1: IDLE1m
;31mE (436245) task_wdt: Print CPU 0 (current core) backtracem
Backtrace: 0x401639FE:0x3FFBE920 0x400833ED:0x3FFBE940 0x4000BFED:0x3FFD9460 0x400951A6:0x3FFD9470 0x40092A95:0x3FFD9490 0x400836BA:0x3FFD94D0 0x4008376D:0x3FFD9500 0x400D5073:0x3FFD9520 0x400F7761:0x3FFD9590 0x401DCE6B:0x3FFD95E0 0x401152FB:0x3FFD9600 0x40115D8D:0x3FFD9870 0x4010A7E4:0x3FFD98B0 0x400F088F:0x3FFD9910 0x400F458A:0x3FFD9930 0x400F4795:0x3FFD99F0 0x400D81B6:0x3FFD9A10
;31mE (436245) task_wdt: Print CPU 1 backtracem
Backtrace: 0x40086F95:0x3FFBEF20 0x400833ED:0x3FFBEF40 0x4000BFED:0x3FFBC7E0 0x400951A6:0x3FFBC7F0 0x40163C6F:0x3FFBC810 0x40163C7B:0x3FFBC840 0x400D418E:0x3FFBC860 0x400933B4:0x3FFBC880
```

backtrace info CPU 0

```c
0x401639fe: task_wdt_isr at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/esp_system/task_wdt.c:183
0x400951a6: vPortClearInterruptMaskFromISR at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:571
0x400951a6: vPortExitCritical at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/freertos/port/xtensa/port.c:332
0x40092a95: xQueueGenericSend at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/freertos/queue.c:946
0x400836ba: lock_release_generic at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/newlib/locks.c:201
0x4008376d: _lock_release at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/newlib/locks.c:207
0x400d5073: esp_vfs_select at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/vfs/vfs.c:918
0x400f7761: net_esp32_recv at /Users/markbecker/Documents/github/Espruino/Espruino/libs/network/esp32/network_esp32.c:251
0x401dce6b: netRecv at /Users/markbecker/Documents/github/Espruino/Espruino/libs/network/network.c:722
0x401152fb: socketClientConnectionsIdle at /Users/markbecker/Documents/github/Espruino/Espruino/libs/network/socketserver.c:676
0x40115d8d: socketIdle at /Users/markbecker/Documents/github/Espruino/Espruino/libs/network/socketserver.c:834
0x4010a7e4: jswrap_net_idle at /Users/markbecker/Documents/github/Espruino/Espruino/libs/network/jswrap_net.c:31
0x400f088f: jswIdle at /Users/markbecker/Documents/github/Espruino/Espruino/gen/jswrapper.c:1907
0x400f458a: jsiIdle at /Users/markbecker/Documents/github/Espruino/Espruino/src/jsinteractive.c:2537
0x400f4795: jsiLoop at /Users/markbecker/Documents/github/Espruino/Espruino/src/jsinteractive.c:2640
0x400d81b6: espruinoTask at /Users/markbecker/Documents/github/Espruino/Espruino/targets/esp32/main.c:108
```

backtrace info CPU 1

```c
0x40086f95: esp_crosscore_isr at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/esp_system/crosscore_int.c:92
0x400951a6: vPortClearInterruptMaskFromISR at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:571
0x400951a6: vPortExitCritical at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/freertos/port/xtensa/port.c:332
0x40163c6f: esp_task_wdt_reset at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/esp_system/task_wdt.c:334
0x40163c7b: idle_hook_cb at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/esp_system/task_wdt.c:80
0x400d418e: esp_vApplicationIdleHook at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/esp_system/freertos_hooks.c:51
0x400933b4: prvIdleTask at /Users/markbecker/Documents/github/Espruino/Espruino/esp-idf-4/esp-idf/components/freertos/tasks.c:4099

```

MaBecker, 2024-12-29:
It does not happen when using tinyMQTT.

sample coding

```js
wifi.disconnect(() => {
  wifi.connect(ssid, {password: password}, function(e) {
    if (e === null ) {
      print("wifi: connected");
      print("mqtt: connect to broker");
      mqtt = require("tinyMQTT").create("test.mosquitto.org");
      mqtt.on('connected', function() {
        print("mqtt: connected");
      });
    } else {
      print("wifi: connect error:",e);
    }
  });
});
```

MaBecker, 2025-01-06:
@gfwilliams  Any clue what is causing this?

gfwilliams, 2025-01-07:
Not exactly, no...

But the way networking usually works in Espruino, the `esp_vfs_select` function in the stack trace (eg https://github.com/espruino/Espruino/blob/master/libs/network/esp32/network_esp32.c#L251) should complete immediately and return whether there's any data available on the channels its called on - that's why it's called with a timeout set to `0`.

If the WDT fails, presumably it's blocking in that function which it shouldn't? It might be worth adding some print statements to check if it is blocking or if it's just chance the backtrace included it.

**But** actually I think in other builds like C3 we might try and disable task_wdt causing a reboot so maybe it's a sdkconfig issue? Even if that's the case we should probably fix the `select` call if it is blocking because that'll be screwing up other execution

MaBecker, 2025-01-10:
Just set `ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0(=n) "Watch CPU0 Idle Task"` and extendet the tinyMQTT to receive messages and is works. But I guess this is just a work around.

gfwilliams, 2026-01-21:
>  Should an idle delay/ stall (eg while(1) ) trigger the watchdog reset , when the E.enableWatchdog(0.5); has been executed?

Yes. that's the idea - if the code hangs it should reboot.

Obviously Wifi stuff may stall a bit, but in that case folks just need to increase the timeout if they want a watchdog

SimonGAndrews, 2026-01-31:
Ok  @gfwilliams .  Having experimented with this,  ive concluded the cleanest way to implement the preferred behaviour is to just not turn on the watchdog timer in the sdkconfig file ( Espruino/targets/esp32/IDF4/sdkconfig_c3)  for the c3 build.   That is CONFIG_ESP_TASK_WDT is NOT set.    I will PR the tested version of the config that achieves this below.   This is the current setup in the original esp32 and the newer esp32 S3 build sdkconfig files.    This appears to be the design  intent of the original ESP32 authors.  

Then without the `wdt` being enabled in the config, the descriptions in the [Espruino API for the wdt](https://www.espruino.com/Reference#l_E_enableWatchdog) apply correctly.   EG on initially running 
```js
while(1)
```
Espruino will hang.    The after implementing the watchdog timer with:
```js
E.enableWatchdog(0.5)
```
`while(1) `  will trigger a reset  . with the log 
```'js
>E.enableWatchdog(0.5)
=undefined
>while(1)
;31mE (66385) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:m
;31mE (66385) task_wdt:  - espruinoTask (CPU 0)m
;31mE (66385) task_wdt: Tasks currently running:m
;31mE (66385) task_wdt: CPU 0: espruinoTaskm
;31mE (66385) task_wdt: Aborting.m
```

and the use of  `E.kickWatchdog()` appears to work also as per the documentation. 

Just for reference; i did try the following and similiar code, unsucessfully,  to synchronise the Espruino `wdt` behaviour  with either sdkconfig,  using the esp-idf `esp_task_wdt-status()` .  It was problematic calling this;  getting both the wdt enabled and regestered to a task,  became very tricky during initialisation,  particularly when using the webIDE with various save on send and other features triggering different soft and hard espruino resets.   Anyway simple is better.  So just for reference (THIS IS NOT IMPLEMENTED)  :
```js
#ifdef CONFIG_ESP_TASK_WDT
static void esp32SyncWatchdogFromIDF(void) {
  if (wdt_enabled) return;
  esp_err_t status = esp_task_wdt_status(NULL);
  if (status == ESP_OK) {
    wdt_enabled = true;
    jsiStatus |= JSIS_WATCHDOG_AUTO;
    ESP_LOGW("Espruino", "ESP-IDF task watchdog active from sdkconfig; auto-kicking enabled");
  } else if (status == ESP_ERR_NOT_FOUND) {
    ESP_LOGW("Espruino",
      "ESP-IDF task watchdog configured but espruinoTask is not registered; use E.enableWatchdog() if needed");
  }
}
#else
static inline void esp32SyncWatchdogFromIDF(void) {}
#endif
```

gfwilliams, 2026-02-02:
Should we also make the change in `targets/esp32/IDF4/sdkconfig`? Not that we use that currently for any builds, but it could be a good plan?

MaBecker, 2026-09-15:
Fixed

closed 2026-09-15 by MaBecker