espruino/Espruino #2589: ESP32 IDF4 Build - I2C.Setup function results in Invalid Arguments on clock speed

SimonGAndrews, 2025-01-03:
An error occurs using espruino in the ESP32-C3 IDF 4 build. when setting up the i2C.   
[image]
In more detail:  When building with the command line: 
`BOARD=ESP32C3_IDF4 DEBUG=1 make flash  `

The error occurs when executing the espruino command: 
`I2C1.setup({scl:D9,sda:D8}); `

 The error text is : 
```
i2c: i2c_param_config(684): i2c clock choice is invalid, please check flag and frequency 
Uncaught Error: jshI2CSetup: Invalid arguments at line 1 col 27 
I2C1.setup({scl:D9,sda:D8}); 
```

SimonGAndrews, 2025-01-03:
This [link](https://esp32.com/viewtopic.php?t=21030) seems to identify the problem. 
And tracing the error message back into the code shows that the ESP IDF i2c driver (home/Espruino/esp-idf-4/esp-idf/components/driver/i2c.c)  function i2c_get_clk_src() , inside i2c_param_config() is expecting to be passed . clk_flags in the config structure from espruino jshardwareI2c.c  
[image]

i2c_get_clk_src() proides [clock selection functionality](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32c3/api-reference/peripherals/i2c.html#source-clock-configuration) in the IDF V4.4 for the specific device..

SimonGAndrews, 2025-01-05:
testing is WIP

gfwilliams, 2025-01-06:
Thanks! This all looks good - if you could get your changes as a PR (or just a patch I can apply) that'd be great.

On the I2C pin front, on STM32 where we have a similar situation of hard-coded pins, we have a checkPinsForDevice function which calls `jshFindPinForFunction(device, functions[i]);` which works out what pin to used based on the auto-generated `jspininfo.c` file: https://github.com/espruino/Espruino/blob/master/targets/stm32/jshardware.c#L2532-L2535

So I think you could do:

```
if (!jshIsPinValid(info->pinSCL)) info->pinSCL = jshFindPinForFunction(device, JSH_I2C_SCL);
if (!jshIsPinValid(info->pinSDA)) info->pinSDA = jshFindPinForFunction(device, JSH_I2C_SDA);
```

Which would work if the pininfo file was correct - I'm not sure it is right now, I think we only do UART: https://github.com/espruino/Espruino/blob/master/boards/ESP32.py#L187-L190

But it would be a nice one to fix, as it's also used for the nice graphics in https://www.espruino.com/ESP32#gpio-pins - for example the one for an STM32 board shows what peripherals are where: https://www.espruino.com/Pico#pinout

gfwilliams, 2025-01-06:
Thanks! I've just pushed changes that add the relevant data for I2C1 - I wasn't too sure about UART/SPI/etc on other boards

gfwilliams, 2025-01-13:
That looks excellent - thanks for your work on this!

SimonGAndrews, 2025-01-13:
** regarding the default I2C2 pins on the ESP32.   I have made sure that on both I2C1 and I2C2 the default pins have not changed from the previous Espruino builds to avoid any breaking changes.  However: 
**for i2C1,** the sdl and scl  pin defaults on the current ESP32 build are actually switched around from the common ESP32 documentation. Espruino has assigned SDA as P22 and SCl as P21 , where has [common documentation](https://docs.espressif.com/projects/arduino-esp32/en/latest/boards/ESP32-DevKitC-1.html#pin-layout) and the [espressif Arduino Library](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/i2c.html#setpins) has SCL as P22 and SDA as P21.  
**and for I2C2,** the existing pins as SDA 17 and scl 16 seem not to be documented anywhere. 

With this in mind I would suggest (And will change in my PR) adding this to the documentation for i2c.setup in jswrap-spi_i2c 
[image]

gfwilliams, 2025-01-13:
> for i2C1, the sdl and scl pin defaults on the current ESP32 build are actually switched around

Have you been able to test? I would assume with ESP32 that what Espruino did was most likely wrong to begin with - so if you can change it to match what everyone else did I think it would be better to do that than to try and soldier on with nonstandard pins

SimonGAndrews, 2025-01-13:
Hi , Yes I Did test, and rechecked today.   Ive checked with an I2C desplay device including monitoring with a logic analyser.   

[image]

[image]

Conclusion as before: On an ESP32 target pre any changes (Espruino 2V24 ) using default (ie no config object passed in I2C.setup() ) , for correct operation a device needs to be wired with SDA on P22  and SCL on 21.  This is opposite to the documentation and pinout diagrams ive found. 
So following your advice I will correct this in the ESP32.py file.

gfwilliams, 2025-01-14:
Wow, ok - thanks! That's crazy. I wonder what's up with that then...

SimonGAndrews, 2025-01-19:
On final testing , I realised that setting the required clock speed was not just a matter of setlecting the correct clock via the . clk_flags (as discussed above).  The IDF also uses the value passed in .master.clk_speed against the selected clock to set the actual speed. 
During testing a logic analyser was used to check the actual speed the I2C was using.

[image]

Speed setting tests using both of the available clocks on the ESP32-C3 were measured using the analyser and it was concludeded to hardcode the selection of the 2MZ XTAL clock by setting  the .clk_flags to Zero.   This gave an accurate clock speed up to 400 hz (or 400000 bitrate setting) .    I was not successful in setting a clock much higher than this accurately.  Maybe my measurment method, wiring pullups etc, or just the device cant do it.  

So the final change in the jshI2CSetup Looked like this.  

[image]

This leaves the ESP32 device I2C speed selection as was before the change. 
Note there are further considerations in the clock selection eg the other clock (1MHz RTC clock ) is documented as being better for low power.  But will leave to others to invistigate and possible add the clg_flags to the espruino i2Csetup function config parameters .  In any event as above these changes will need further update for IDF V5.

SimonGAndrews, 2025-01-19:
Finally all builds were sucessfull in the github build workflow of my Espruino clone.  So going to raise my first PR with these changes. :)

gfwilliams, 2025-01-20:
Just merged - thanks for all your work on this! It's great that you were able to check the speeds with the logic analyser too!

closed 2025-01-20 by gfwilliams

SimonGAndrews, 2025-01-28:
Just as a follow up , further testing while on issue #2601 showed that the ESP32S3 build did have an issue with a default setup on I2C2 ie I2C2.setup().   

This was because the ESP32C3_IDF.py board file did not assign default pins for I2C2.  This was fixed in the PR raised for 2601.