espruino/Espruino #2597: [Feature request] RMT driver.

0wwafa, 2025-01-15:
I thought it was implemented but it's not :(

```
var rmt = require("rmt");

var rmt_rx = rmt.create({
  pin: D16,
  channel: 0,
  clkDiv: 80, // 80MHz clock divided by 80 gives 1MHz (1us resolution)
  memBlocks: 1,
  filter: true,
  filterThreshold: 100,
  idleThreshold: 0xFFFF,
  rx: true
});
```

0wwafa, 2025-01-15:
RMT is extremely useful not only for infrared transmit/receive but to accurately read the length of PWM pulses up to a resolution of 12.5 nanoseconds (80 Mhz, divisor =1)

gfwilliams, 2025-01-16:
Is this for ESP32 or some other board?

You're welcome to contribute code to do this if you want - but Espruino is Open Source, we make zero money out of ESP32 and Espressif provides us with no support, so this isn't something that's likely to be implemented unless you're willing to do it.

0wwafa, 2025-01-17:
ESP32:
In micropython it's implemented only for sending.
in circuitpython is not implemented at all.
It's a shame because it's one of the coolest features!

[image]