espruino/Espruino #2599: Arc Algorithm based on Bresenham Circle routine

MaBecker, 2025-01-16:
This coding looks very promising to me,  What do you think @gfwilliams ?

source: https://www.scattergood.io/arc-drawing-algorithm/
formated and copied to https://gist.github.com/MaBecker/071ca5440565dc9163a8e35a0cf2c5fc

gfwilliams, 2025-01-16:
Thanks - what's the use case for this out of interest?

My concern is that if it goes in, it will only be able to be for some boards - some of them like the original Espruino that are too tight on flash storage will have to leave it out.

... And we do already have drawEllipse: https://github.com/espruino/Espruino/blob/master/libs/graphics/graphics.c#L532

I think it got mentioned in a post on the forum, but if difference between start and end angles is less than 180 degrees it should be possible to implement it just using drawEllipse (which I believe uses the same algorithm) and then just setting the clipRect, which feels like it would only add 100 bytes max and could probably be kept in for all boards...

MaBecker, 2025-01-16:
there is no use case at the moment. Was interested in how to draw arcs and because of this post: https://forum.espruino.com/comments/16966259/
I will work on this and let you know about the size.

MaBecker, 2026-04-25:
Saw some coding Gordon implemented something similar - so let's close this

closed 2026-04-25 by MaBecker