espruino/Espruino #2601: ESP32 - S3 build (ESP IDF 4) - Error Initialising SPI1

SimonGAndrews, 2025-01-23:
See [Forum](https://forum.espruino.com/conversations/402116/) Where issues identified by @devsnd result in a restart of the ESP32 - S3 when  
`SPI1.setup({ sck:D4, mosi:D5 });` 
is executed in a script.   
The issue does not occur when this code is used:
`var spi = new SPI();   
spi.setup({ sck:D4, mosi:D5 });
`

The restart is preceeded with the message:

[image]

spi_bus_initialize is called in Espruino from targets/esp32/jshardwareSpi.c  
Investigation to follow.

MaBecker, 2025-01-23:
My guess: This could be some hard coded  dma channel or an sdconfig issue.

SimonGAndrews, 2025-01-24:
Thanks, yep will check out  the use of SPI_DMA_CH_AUTO
And look at the pins.  The i2c2 pins need defining also.  I’m not as familiar with SPI so doing a little home work this am.   Should be able to sort.

gfwilliams, 2025-01-24:
> I’m not as familiar with SPI so doing a little home work this am

Thanks! If you want a quick test, connect MOSI and MISO pins together, and you should then be able to do `SPI.send("Hello")` and get the same characters back!

SimonGAndrews, 2025-01-24:
Nice Tip.  Thanks.
Just to confirm this issue also occurs in the ESP32C3_IDF4  (ESP32-C3 target) build so will fix for both targets.

SimonGAndrews, 2025-01-27:
Using  SPI_DMA_CH_AUTO seems to be the answer.  Testing under way including making sure ESP32 still works.

SimonGAndrews, 2025-01-28:
The Change to fix the issue was to use the SPI_DMA_CH_AUTO flag in the DMA parameter in the ESPIDF spi_bus_initialize () function in  file /targets/ESP32/jshardwareSpi.c
This file also used hardcoded SPI pin and SPI Peripherial Host IDS which were limited to the ESP32 and did not include those for the ESP32C3 and ESP32S3.   Changes to accomodate the additional boards and the hardcoding were made by using Pin configurations in the relevent Boards.py file for the targets.

SimonGAndrews, 2025-01-28:
The board files ESP32.Py , ESP32C3_IDF4.py and ESP32S3_IDF4 were modified as in this example for the S3.

[image]

The choice of pins  was made considering:
-  the IDF files in  /components/soc/Target/.../pins.h  These target specific files in the ESPIDF identify peripherial pins which can be mapped directly through the GPIO system in support of high-speed digital signals such as SPI ; bypassing the GPIO matrix for better high-frequency digital performance.  See the GPIO reference section [here](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf).   These also appear to correspond to commonly documented defaults.
- ESP32 [Reference](https://docs.espressif.com/projects/esp-idf/en/v3.3.4/api-reference/peripherals/spi_master.html?highlight=spi#gpio-matrix-and-iomux)

SimonGAndrews, 2025-01-28:
The essential new code  is 

[image]

[image]

SimonGAndrews, 2025-01-28:
Testing of the change was carried out with Debug builds for the ESP32 in IDF V3 and ESP32C3 and ESP32S3  in IDF V4 as the one jshardwareSPi.c file is used for all three.  The new debug message showed allocation of host devices and pins as expected from the new board files.
Loop back testing of MISO pins connected to MOSI pins confirmed the SPI as working with SPIn.send() JS functions and not in the case where not connected. 

[image]

[image]

Build completion of all three targets was checked using the actions build.yml workflow in a local clone of the Espruino Github Repository  (modified with DEBUG=1)

SimonGAndrews, 2025-01-28:
Notes on additional testing:

[image]

Interesting so see the IDF on the S3 setting pin parameters.  This did not seem to happen on the ESP32 build in IDF V3.  And using an invalid Pin (eg an input only pin) on the ESP32 does result in an untrapped restart.

[image]

gfwilliams, 2025-01-29:
Thanks for all your work on this, it makes a huge difference having you putting so much time into the ESP32 ports - all merged in now!

gfwilliams, 2025-01-29:
I just wanted to add that your mention of the "GPIO matrix" is something I never knew about ESP32! So I guess it explains why things still worked even though the pins weren't correct - it was using the matrix so it worked but just wasn't as fast as it could have been if going direct.

closed 2025-01-29 by gfwilliams

SimonGAndrews, 2025-01-29:
> Thanks for all your work on this, it makes a huge difference having you putting so much time into the ESP32 ports - all merged in now!

Your very welcome , Espruino is a great product :) , its well worth the effort.

MaBecker, 2025-01-30:
Just checked for ESP32_CYD_IDF4_CUSTOM board

```js
>ESP32.getState()
={
  sdkVersion: "v4.4.8",
  freeHeap: 39828, BLE: false, Wifi: true, minHeap: 37480 }
>process.env
={
  VERSION: "2v25.58",
  GIT_COMMIT: "944256c77",
  BOARD: "ESP32_CYD_IDF4_CUSTOM",
  RAM: 524288, FLASH: 0, STORAGE: 262144,
  SERIAL: "f8b3b72b-a62c",
  CONSOLE: "Serial1",
  MODULES: "Flash,Storage,hea" ... "lnetServer,crypto",
  EXPTR: 1061219648 }
```

@SimonGAndrews  & @gfwilliams  Many Thanks