espruino/Espruino #2602: Looks like g.fillPoly() is missing a line

MaBecker, 2025-01-23:
[image]

looks like bottom line is missing for the right box.

```js
// like rectangle
function polyRectFase(x1,y1,x2,y2,f) {
  return[ x1,y1+f, x1+f,y1, x2-f,y1, x2,y1+f, x2,y2-f, x2-f,y2, x1+f,y2, x1,y2-f];
}

g.clear().setColor("#f00");

// check boxes
g.drawPoly(polyRectFase(60,90,80,110,3),1).fillPoly(polyRectFase(90,90,110,110,3),1);

```

@gfwilliams Hope this is not to complicate to fix.

MaBecker, 2025-01-23:
[image]

another sample 

```js
g.drawPoly(polyRectFase(60,90,80,110,3),1).drawPoly(polyRectFase(90,90,110,110,3),1)
 .drawPoly([61+5,94+4, 61+3,94+6, 61+8,94+11, 61+15,94+4,  61+13,94+2, 61+8,94+7],1)
 .setColor("#f00").fillPoly([91+5,94+4, 91+3,94+6, 91+8,94+11, 91+15,94+4,  91+13,94+2, 91+8,94+7],1);

```

MaBecker, 2025-01-24:
workaround for now: use draw and fill

```js
g.clear().setColor("#f00")
 .drawPoly([5,4, 3,6, 8,11, 15,4, 13,2, 8,7],1)
 .fillPoly([5,4, 3,6, 8,11, 15,4, 13,2, 8,7],1);
```

[image]

gfwilliams, 2025-01-24:
Looking at the docs:

https://www.espruino.com/Reference#l_Graphics_fillPoly

> This fills from the top left hand side of the polygon (low X, low Y) down to but not including the bottom right. When placed together polygons will align perfectly without overdraw - but this will not fill the same pixels as drawPoly (drawing a line around the edge of the polygon).

Is this not kind of what you're seeing here? It may seem like a strange thing to do but I believe this is pretty standard for a fillPoly in computer graphics, and I seem to remember having to go through a few iterations to get to this (see [the blame log](https://github.com/espruino/Espruino/blame/1dccab89fa8fcf5b6ca6bb0af487e03d6778e45b/libs/graphics/graphics.c) for fillPoly). One of the big drivers for this is Vector fonts, where we may make a character with more than one fillPoly - if the filled polygon extends too far then the fonts end up being over-filled, and especially when they're small they are unreadable.

MaBecker, 2025-01-24:
Ok, so it  works as implemented .

closed 2025-01-24 by MaBecker