espruino/Espruino #2606: General I2C questions

MaBecker, 2025-02-01:
On my way to implement a CST816S i2c touch device for a custom board.

Is there a implementation like `function I2C.readReg(address, reg, quantity)` we can use in c instead of a  `jsi2cWrite()` used with `jsi2cRead()`

```
#define CST816S_ID 0x15

unsigned char buf[6];
uint8_t version;
// read version
    buf[0] = 0x15; jsi2cWrite(&jswrap_touch_init, CST816S_ID, 1, buf, true);
    // wait 100ms
    jshDelayMicroseconds(100*1000); // we should really have a callback
    jsi2cRead(&jswrap_touch_init, CST816S_ID, 1, &version, true);

```

SimonGAndrews, 2025-02-01:
There may be some ideas here.
[This link](https://www.espruino.com/PCA9685)
And the supporting module http://www.espruino.com/modules/PCA9685.js have some examples of setting up a device via its registers.  That may help.

SimonGAndrews, 2025-02-01:
But your probably familiar with those @MaBecker ??

MaBecker, 2025-02-01:
Thanks for picking this up. Yes
This a question about firmware code in c.

MaBecker, 2026-10-16:
Device needs a rest like used in c code for bangle

https://github.com/espruino/Espruino/blob/2735c32d478068d8e25be26870590b5e4121699b/libs/banglejs/jswrap_bangle.c#L3978-L3983

closed 2026-10-16 by MaBecker