espruino/Espruino #2607: ESP32S3 - I2C issues

MaBecker, 2025-02-02:
Sample code to read version from a CST816S i2c module from a [ESP32-S3-Touch-LCD-1.28](https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.28#Overview) device

```js

// CST816S
const ID = 0x15, REG_VERSION = 0x15, REG_VERSION_INFO = 0xA7;

I2C1.setup({sda:D6,scl:D7});

print('Version:', I2C1.readReg(ID, REG_VERSION, 1));
print('Version Info:', I2C1.readReg(ID, REG_VERSION_INFO, 3));
/*
Uncaught Error: jshI2CWrite: slave doesn't ACK the transfer
 at line 39 col 50
print('Version:', I2C1.readReg(ID, REG_VERSION, 1));
                                                 ^
Uncaught Error: jshI2CWrite: slave doesn't ACK the transfer
 at line 40 col 60
...Reg(ID, REG_VERSION_INFO, 3));
                              ^
*/
```

MaBecker, 2025-02-02:
@SimonGAndrews  Which module i2c module did you tested with an ESP32-S3?

SimonGAndrews, 2025-02-02:
I was Using a ESP32-S3-WROOM-1-N8R8 on an
Olimex ESP32-S3-DevKit-Lipo Development Board.  
[Here](https://thepihut.com/products/olimex-esp32-s3-devkit-lipo-development-board?variant=43878742786243&country=GB&currency=GBP&utm_medium=product_sync&utm_source=google&utm_content=sag_organic&utm_campaign=sag_organic&gad_source=1&gclid=CjwKCAiAzPy8BhBoEiwAbnM9OyoNY6aW7oKDUsjPMJAGfmOCjmymTvq8nSkedZMFZwXRjnxGKiX2eBoCWbAQAvD_BwE)

MaBecker, 2025-02-02:
@SimonGAndrews    Which i2c device did you use for testing?

gfwilliams, 2025-02-03:
You could check out what mentioned @SimonGAndrews in https://github.com/espruino/Espruino/issues/2589 - looks like it was an SSD1306 display.

@MaBecker it's an on-board I2C thing you're using is it? Can you access it using software I2C? Just to rule other issues out?

I wonder whether maybe there aren't pullup resistors but it expects the chip to put its' internal ones on?

@MaBecker maybe you could try some other hardware on the S3 and see if that works?

SimonGAndrews, 2025-02-03:
@MaBecker Good morning,  Yes I agree something still not correct with the S3 I2C.
Im testing with a PCA9685 on ESP32 and C3 .    Imusing one of the off the shelf servo/led x 16 driver boards with leds soldered on the outputs.      However, on the S3 i did not get as far as using actually with this device, I stopped after setup gave me the correct debug response at setup  (ok sorry lesson learnt)   and on checking this am there does seem to be some issues (not exacly like yours, but time outs on writting to the device) .  So it doesnt look right on first attempt.   i will setup a logic analyser on it and check the S3 technical Spec for any differences in the I2C section.

PS might be worth building with DEBUG=1 as the code I changed put out a few debug messages onsetup etc.

SimonGAndrews, 2025-02-03:
[image]
S3 test setup

SimonGAndrews, 2025-02-03:
[image]

SimonGAndrews, 2025-02-03:
OK , @MaBecker  I think my initial issue,  was with the pin I was trying , on my board, for sda (D6 as in your case) was tied to ground for a battery status facility.   This error was a timeout error.

On changing the pin to D15 I got the above result , Ive not checked the data but the readReg() function does appear to work.    

I did get your error when I used the wrong address of the I2C.

SimonGAndrews, 2025-02-03:
[image]

MaBecker, 2026-09-15:
Have to check again with a proper setup

MaBecker, 2026-10-07:
@gfwilliams is the a CST816S implementation? Would make my tests easier.

gfwilliams, 2026-10-07:
It's built in to Bangle.js 2 (jswrap_bangle) - overall it's very simple, you just have to reset it to start it up

closed 2026-10-07 by MaBecker