espruino/Espruino #2609: ESP32 S3 - USB Serial JTAG Console not Support in current code

SimonGAndrews, 2025-02-06:
There seems to be an issue in that the ESP32S3 build does not include support for the USB Serial JTAG Console.   That is boards where the USB connector (for connection to the host/IDE) is routed directly to the processor (GPIO 19/ D+  and 20 / D- ) and not  through a USB to Serial Convert chip to 43/U0TXD  and  44/ U0RXD.

The symptom is no response in the IDE after flashing.  The espressif monitor confirms flashing is ok and the espruino banner displays after a reboot, but then no console/repl.

Ive seen other [forum posts](https://forum.espruino.com/conversations/372053/?offset=25#comment17609505) that indicate the same.

In the current code ../targets/ESP32/jshardwareUart.c  handles this for the ESP32C3 but the #ifdef in the code does not include the S3.   Ive thad a go to fix in threee places:

[image]
[image]
[image]

Which has improved in that I get a console now.  But it now has the problem seen during the initial port where there is no echo till enter key is pressed.  

I Also note the need to define USB_CDC in board.py,  whcih ive done. And it looks to be set in the IDE.  

[image]

WIll keep going to figure it out, but I cant initially see where to fix the Echo, given it was fixed in the C3.

gfwilliams, 2025-02-06:
Thanks for looking into this! IIRC Espruino was getting the data as written and processing it, but the data was only being sent out to USB after a newline. I'm pretty sure this was the code, so hopefully a check for ESP32S3 in there too would fix it:

https://github.com/espruino/Espruino/blob/master/targets/esp32/jshardware.c#L595-L599

gfwilliams, 2025-02-06:
I think you need it here too: https://github.com/espruino/Espruino/blob/master/targets/esp32/main.c#L47-L51

SimonGAndrews, 2025-02-06:
Sorry one more consideration:

does @MaBecker , @gfwilliams or anyone object to changing  USB_CDC to   USB_SERIAL_JTAG 
In board.py and throughout.
Arguably  more intutive and but is  in line with the ESPIDF config whcih provides these thre config parameters 
```
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG
CONFIG_ESP_CONSOLE_USB_CDC 
CONFIG_ESP_CONSOLE_UART
```

and then our condition becomes 
`#if (ESP_IDF_VERSION_MAJOR >= 4) && defined(USB_SERIAL_JTAG)`

Im not suggesting to use these  ESPIDF configs now , we stick witth the def in board.py ,  But see this example for how we could look in the future.

[image]

gfwilliams, 2025-02-06:
> #if (ESP_IDF_VERSION_MAJOR >= 4) && defined(USB_CDC)

Sounds good to me. Is this IDF4+ specific, or is it for devices with USB built in though? Eg would it work for `ESP32_IDF4` where it's just an ESP32 on IDF4? Might be worth having something in platform_config (built with build_platform_config.py) which was just `ESPR_SERIAL_JTAG` or something?

> anyone object to changing USB_CDC to USB_SERIAL_JTAG

I'm fine to change it, but personally I'd like to change the polarity to something like `ESP_FORCE_NO_USB_SERIAL` or similar.

I feel like the vast majority of ESP32C3/S3/etc are going to have serial over USB - that's one of the great things of the chip. It should really be opt-out IMO

gfwilliams, 2025-02-06:
That looks excellent, thanks! Way better being able to use something SDK-defined!

SimonGAndrews, 2025-02-06:
@MaBecker , hi are you ok with this approach?

MaBecker, 2025-02-06:
Yes, go - thanks for your effort

SimonGAndrews, 2025-02-12:
Having tested changing parameters in the SDKConfigs to identify target boards uisng the ESP_CONSOLE_USB_SERIAL_JTAG im not going to include that in the solution. The available SDKCONFIG parameters were changing the behaviour of the console USB in a way I couldnt fully understand.  It was fine for the simplier boards but di not work well for the more complex (I will detail these in testing notes to follow)

Instead im proposing to use a macro in the targets/ESP32/JShardwareUart.h like this 

[image]

and a test condition in the the code 
`#if TARGET_PROVIDES_USB_SERIAL_JTAG && !defined(ESP_FORCE_NO_USB_SERIAL_JTAG) `

It does mean the code will need to be modified if another board comes along with the functionality.  But it will only need changing in the one place.   
 
The Boardfile.py changes as follows are working fine:

[image]

Testing notes and PR to follow

SimonGAndrews, 2025-02-13:
il will test this:

[image]

SimonGAndrews, 2025-02-13:
Thanks , i will leave the port setting alone.

I didnt have to change any includes , JShardwareUart.h  , was included in all the files that needed the condition.

gfwilliams, 2025-02-13:
> I didnt have to change any includes , JShardwareUart.h ,

Sounds great then!

gfwilliams, 2025-03-03:
How's this looking? I see you've made a few commits (on your branch?). Do you think they'd be ready for a PR to get this fixed?

SimonGAndrews, 2025-03-03:
Apologies for the delay.  the fix is ready for a final test on the S3/C3 and ESP32 to make sure its still ok.   I recommited my changes to my branch tonight becuse there were other changes in the same files during my delay.  Ive unpicked them and my branch is synced except the  PR with my changes ready to go after testing.  The build workflows all completed in my branch tonight.  pretty confident with it but would prefer to test again.   Sorry but im on holiday till march 17 so can test then.   Happy to send the PR if someone else wants to test.

gfwilliams, 2025-03-04:
Ok, great - thanks!

No rush at all (unless as you say someone else really wants to test this) - hope you have a nice holiday!

SimonGAndrews, 2025-03-19:
Quick Status  @gfwilliams .   I have a tested version that gets the s3 with jtag USB working and regression is ok with ESP32 and ESP32C3.   However there are a couple of test cases that are failing where board provides aditional USB or ESP prog port wired to the uart .  My plan was these would work with the Serial override flag set.    One of the boards is the Espressif Devkit  that @rgomezwap was using so would like to see that work.    Im pusing on with fixing that .  But if you want an interimum PR with that still as an issue shout.   
PS  there is a new issue in the ESP32S3_IDF4.py file after the last commit.   
 pins = pinutils.generate_pins(0,39)    Needs updating to 48 pins give the new default on pin 43 for USART1_TX.

SimonGAndrews, 2025-03-19:
Also just for consideration ,  do we have a gap in the approach to adding the USB via JTAG as the console.   E.getConsole  shows the console as 'Serial1'  , as one might expect with the current approach.     Whats involved in adding a USB device to board file and using the JTAG driver in the case of the USB being the default console,  maybe next setep is sort setConsole and GetConsole to properly flip between USB and Serial1 uart ?    One area I dont fully understand is the Device handling in the IO Event queue , is that necessary or can we just use similiar to the exisitng methods in main.c and jshardwareuart.c of flipping to the USB jtag or UART .    But utilising the jsiGetConsoleDevice()  and similiar functions ?

SimonGAndrews, 2025-03-20:
Im going to test a version of this fix which uses the setting of default_console in the board.py file (and including a USB device)  to set USBSERIAL as the default.   In theory it should allow the E.SetConsole() function to change to Serial on the fly whcih would be usefull with the dual USB port boards.   

I think ive traced through the setup of the serial devices device down to initSerial()  in jshardwareuart.c and I ve modified to include EV_USBSERIAL (See pic) and put the Jtag serial initialisations there.   

The consoleToEspruino()  function can then use the jsiGetConsoleDevice() function to select the correct serial read function (pic 3).   Ive kept the flushing bits for the USB jtag.   I had to hack the is USB connected function as there is not an available espidf call until v5. 

I may not get it right but I think its fundementally the right approach to suit the underlying Espruino console support.   Its not building yet!  and Testing is next.

[image]
[image]
[image]

gfwilliams, 2025-03-20:
Sorry for the delay - yes using EV_USBSERIAL for the built-in USB support sounds like a great idea. As-is it feels like it gets in the way of using `Serial1`?

What'd be ideal is if there was a way of detecting if the USB JTAG was connected. Maybe use `bool usb_serial_jtag_is_connected(void);` in `jshIsUSBSERIALConnected` and then add what we do for STM32 into the ESP32 jshIdle handler:

https://github.com/espruino/Espruino/blob/d2cbb39e4ee6ca7050f93e5cf82450a760ce50e8/targets/stm32/jshardware.c#L1558-L1565

So then hopefully if we're connected by USB JTAG it'll swap automatically, so on boards with it available it'll swap over automatically, but otherwise it'll use whatever the default console device is (Serial1?) on those other devices?

> pins = pinutils.generate_pins(0,39) Needs updating to 48 pins

Oh yes, thanks! I just put that in.

SimonGAndrews, 2025-03-21:
Thanks @gfwilliams , Making some progress.   Having introduced USB as a device in the board file,  It took me a while to  understand the importance of  the function jsiOneSecondAfterStartup() to move the console device out of 'Limbo' when USB is defined.  This function is not currently executed in the ESP32 build and hence console was  switching to Limbo and staying there.   Im adding the execution of the function in main. .   So I have a working build with USB defined  in the board file now!

Regarding 

> usb_serial_jtag_is_connected(void); 

 unfortunately its not available in the current ESPIDF V4 were using.  Its in V5.   WIll try to get a solution without this or a hack for now and the dynamic switching over as in STM example could follow in V5.  Ive made a note in the function jshIsUSBSERIALConnected as todo for V5

If you have a minute to explain  the concept behind jshPushIOCharEvents, KickUart and jshTransmit, I would appreciate it.  
Thanks Again

gfwilliams, 2025-03-21:
Ahh, great! Thanks for spotting the `jsiOneSecondAfterStartup` issue - this all looks really promising.

'EV_LIMBO' is there so we can start running JS code at boot before the USB/whatever has properly initialised without losing data sent to the port - because at startup we don't want  to sent any data to Serial if the board is actually going to end up connected to USB.

> usb_serial_jtag_is_connected(void) unfortunately its not available in the current ESPIDF V4

Argh, that's a pain. I guess worst case we could detect when a character arrives on USB JTAG and use that as a queue to switch over?

> If you have a minute to explain the concept behind jshPushIOCharEvents, KickUart and jshTransmit, I would appreciate it.

Sure! There are two FIFOs in Espruino - one of them (`txBuffer`) is for transmission - code can push stuff into it (`jshTransmit`), and then interrupt handlers generally take it out - for instance an interrupt handler for Serial1 might take characters out of txBuffer whenever it received a 'buffer empty' interrupt. However suppose we just wanted to start transmitting on Serial and we weren't before - in that case we need to start the Serial peripheral transmitting - and that's where `KickUart` gets called usually.

However under an RTOS like the ESP32 I believe there are already output buffers for each peripheral, so potentially there's something to be said for not ever putting data into `txBuffer` - you could add something to `jshTransmit` which would just push the data direct to the RTOS.

The second buffer is `ioBuffer` and that comes *from* Interrupts *to* the event loop (outside of interrupts). So any Serial character received, or pin changing state, or Bluetooth message gets shoved in there. `jshPushIOCharEvents` is what you'd call to to put data in when serial data is received.

... so on ESP32/ESP8266 it's less than ideal as Espruino has these buffers which were designed for very low-level use when running bare metal on microcontrollers, and we're effectively using those *and* ESP32's RTOS ones as well. I think when getting data from Serial devices, `jshPushIOCharEvents/ioBuffer` is probably still the right way - but when sending data `txBuffer` doesn't make that much sense.

SimonGAndrews, 2025-03-22:
With regard to regression testing , changes in the console device setup:
I picked up on @GaryOtt other discussions that he connected  a REPL client to the C3 via bluetooth.  Which to be honest I had not realised was supported in the ESP32 family.  I was equally pleased to hear it worked :).    Ive since quickly tried this on the current released code for C3,  S3 and ESP32  and they all work,  Which is great.
So these will all need a regression test. 
My question is other than these bluetooth , the new SerialUSB  Jtag, and the UART tests , is there any other console setups that need checking; eg Wifi?

SimonGAndrews, 2025-03-23:
Thanks @Gordon for the details above,  ive had a good look at this over the weekend and your start helped me understand alot more.  And I believe I can see the way through to fix the issue by setting up the USB for Jtag on the C3/S3.  
But Sorry one more background question.   
Im Ok with JshUSARTKick() :  It looks like the main route to a console write from txbuffer is via jshTransmit() whcih calls jshUSARTKick().   JshUSARKick() for the ESP32 seems that it is actually pulling the character out of the txbuffer one by one until the buffer is empty on each call.  It has a switch on the device name and handles each device seperately.  For the fix it needs USB case adding to the Switch cases , which should be straignt forward following the existing pattern of just puling the chars out.  So im assumig to add a usb_serial_jtag_write_bytes.  **
[image]

The bit I dont get is whats happening  in jshTransmit() in the case of Bluetooth and USB.  It seems jshTransmitClearDevice() is being called in both cases to discard any data in the txBuffer and then just return (no call to  JshUSARTKick() ) .  I assume it must transfer out of txbuffer by some other mechanism but whats the reason behind this bit of code? 

[image]
[image]

Thanks and sorry for so many questions.  

** I think I also get  your points above on pushing direct to the RTOS instead of the TXbuffer but will leave that for another day.  Im getting lost at the jshTransmit() level and above, as you can see.

SimonGAndrews, 2025-03-24:
Thank you @gfwilliams , Ive got it now.  Ill push on with the fix and tests.

SimonGAndrews, 2025-04-09:
Sorry @gfwilliams Im still working / testing this and have another question regarding the use of "Limbo" .  
Looking at the code in jsiOneSecondAfterStartup.:

[image]

So I assumed that the txbuffer content, with a device flag of 'limbo', was being preserved while in limbo so that the  jshTransmitMove(EV_LIMBO, consoleDevice)  and subsequent USARTKick , in OneSecondAfterStartup, would then transmit all that was sent while in limbo.   But during testing I found  jshTransmit() was being called with the device  limbo to transmit a load of stuff out.   I forced a stack dumb in this case (next pic)  and got what follows:

[image]

[image]

So it looks like in the case of #ifdef USB , jsiInit() having set consoleDevice == EV_LIMBO is then actually transmitting out of the txbuffer.   Am I missing something ?   

As it stands im thinking to ignore the limbo concept and  flip consoleDevice back to DEFAULTCONSOLE, ASAP  Here:
``` 
/**
 * Re-init the ESP32 after a soft-reset
 */
void jshSoftInit() {
 // jsiSetConsoleDevice(DEFAULT_CONSOLE_DEVICE,false);  // SGA TODO Comment here
  if(ESP32_Get_NVS_Status(ESP_NETWORK_WIFI)) jswrap_esp32_wifi_soft_init();
}
```
which is convieniently called in  jsiSemiInit() in the case of #ifdef ESP32,  after JSInit()  sets console to limbo but before any serious jsiConsolePrintf()'s in the whole initialisation.   
I will carry on.

SimonGAndrews, 2025-04-10:
Sorry I should have started with my issue is that I was seeing jshUSARTKick being called with device EV_LIMBO .   Its not so much that jshTransmit was being called with it. I suppose jshTransmit is the best place to trap and handle any transmission while in Limbo.    My question is , I do I understand this correctly,  in that calls to jshUSART with device limbo should not be happenning.

SimonGAndrews, 2025-04-10:
@gfwilliams sorry to drag you into this. And to save on iterations.   Having considered further , unless I’ve got wrong end of stick, my proposal is to modify jsiinit()
`#if defined(LINUX) || !defined(USB)
  consoleDevice = jsiGetPreferredConsoleDevice();
#else
  consoleDevice = EV_LIMBO;
#endif `
So that setting consoleDevice = EV_Limbo does not apply #ifdef esp32 and esp32 build continues to not have that state with the mod to add USB devices

gfwilliams, 2025-04-11:
Thanks for looking into this.

> My question is , I do I understand this correctly, in that calls to jshUSART with device limbo should not be happening.

I feel like it's fine to call jshUSARTKick on EV_LIMBO, but jshUSARTKick **should** just do nothing in that case (so they shouldn't take data out and try and transmit it).

Is there a case in ESP32 where data sent to USB immediately could get lost if we send it out too quickly? I guess not - so if that's the case we could do the `defined(ESP32)` as you say...

If jshUSARTKick on EV_LIMBO happens on other platforms it's something I'll have to fix myself anyway though

gfwilliams, 2026-01-05:
Hi! I can't seem to find the branch you had for this now, but I did see these two:

https://github.com/SimonGAndrews/Espruino/commit/3837fec9d97b15b2e990a987a877165e82d21d43
https://github.com/SimonGAndrews/Espruino/commit/ff664cca537ddefb4f20a547d6019c593b95f5f5

Do you think it'd make sense for me to just merge those in? It feels like there's not  too much to worry about there and it might fix the main issues folks are having with S3

gfwilliams, 2026-01-05:
Thanks! I'll take a look in a few days! I'll make sure I get an ESP32S3 here to have a play with first though :)

SimonGAndrews, 2026-01-15:
@gfwilliams , In case youve not started to look at this yet.
I reviewd the old commits you identified and think that solution was not whole.  The logic in those commits , relied on changes to the sdkconfigs only for builds that were to enable the jtag serial.  Which is not ideal

So Ive created a new branch in my espruino github repro https://github.com/SimonGAndrews/Espruino/tree/fix-for-%232609-V2 .  I think its the simplest change to get the s3 included in working logic. ( Given there is not an easy way in ESP-IDF 4.4.8 to identify if the USB is connected to JTAG or not (it is in V5). )  

It looks whole to me, but i have not tested yet.   The change in summary.

Board definition files can now set ESP_USE_USB_SERIAL_JTAG  (replaces use of Old USB_CDC identifier)

- Boards that have native JTAG USB set it. (eg C3 - S3) 
- Any UART‑bridged boards don’t set it
- Code checks #ifdef ESP_USE_USB_SERIAL_JTAG and uses USB Serial JTAG only then.
- ESP32 (original) never defines it → unchanged.
- see comments in S3 and c3 boards.py

This approach does not mesh with all your feedbak comments, So please dig deeper if you want.  Else I know this needs closing out so Feel free to grab this branch as is (or i can PR IT if you say the word).  Or ill do my best to test in the comming days.  I do have some time on my hands at the moment.

gfwilliams, 2026-01-16:
> boards/ESP32S3_IDF4.py currently has DEFINES+=-DESPR_USE_USB_SERIAL_JTAG commented out

Argh - sorry - that must have been left from my testing. Just fixed!

And I just updated the 'Curio' board which I believe was the only other ESP32C3/S3 in the repo

gwilliams, 2026-01-16:
> You’re most welcome. @gwilliams

Thank you. But you tagged the wrong person 🙂