espruino/Espruino #2612: analogRead() result is always <1, never 1 (at least for for nrf5x and esp32)

fanoush, 2025-02-24:
Not sure what is the original idea behind this and whether result 1 is wanted or not, but looks like for STM32 this can actually return 1

https://github.com/espruino/Espruino/blob/555e338ce9681e116178bac31b48bc83e2938c9a/targets/stm32/jshardware.c#L1903C66-L1903C71

but for both ESP32 and nrf52 this is divided by 16384 or 4096 so the result is always below 1 as the maximum value returned by ADC is less by one
https://github.com/espruino/Espruino/blob/555e338ce9681e116178bac31b48bc83e2938c9a/targets/nrf5x/jshardware.c#L1425
https://github.com/espruino/Espruino/blob/555e338ce9681e116178bac31b48bc83e2938c9a/targets/esp32/jshardware.c#L395

And BTW in case of changing existing constant beware of few lines like this https://github.com/espruino/Espruino/blob/555e338ce9681e116178bac31b48bc83e2938c9a/targets/nrf5x/jshardware.c#L1327C43-L1327C49 which currently results in analogRead returning -1(?)

gfwilliams, 2025-02-24:
Thanks - I feel like it's probably STM32 we should change... It feels like it makes more sense to use a whole number for it - converting to/from floating point is likely to be a whole lot less lossy if so?

The original idea was just that Arduino's analogRead returned values the range of which changed depending on which MCU it was running on, and I wanted to avoid that. It didn't really go much further than that...

fanoush, 2025-02-24:
> The original idea was just that Arduino's analogRead returned values the range of which changed depending on which MCU it was running on, and I wanted to avoid that.

Yes, got that, but was not sure about the maximum - is it supposed to go to 1.0 or should be always less than 1.0.

Because for STM32 it looks like it goes up to 1.0 as the value is divided by 65535 (not 65536) but for nrf5x and esp32 it should be divided by 16383 and 4095 to go fully up to 1.0. Currently it never reaches 1.0 even with maximum value received from ADC.

>  It feels like it makes more sense to use a whole number for it - converting to/from floating point is likely to be a whole lot less lossy if so?

I don't get this. You mean doing breaking change and abandoning the current  idea of analogRead() with converting to float interval 0.0-1.0 altogether and be like Arduino?

gfwilliams, 2025-02-24:
> It feels like it makes more sense to use a whole number for it 

Sorry, I meant "whole power of 2" - so 65536 for STM32 and not 65535

Because looking at it, STM32 left-aligned pads with zeros so even dividing by 65535 wouldn't mean you ever get as high as 1.0 either!

[image]

fanoush, 2025-02-24:
Oh, OK, didn't actually try on STM32 so that is even worse there :-) Because I would see some advantage of actually getting the 1.0 value when we already map to  some artificial interval and do the floating point conversion anyway - so if mapped to some progress bar or something like that it would actually go to 100%

Whether with double precision floats division by 4095.0 or 4096.0 would make difference in precision I don't know.  I would guess that both are imprecise in same way but don't really know. And since the measured analog value is inexact anyway I am not sure it matters, having it go to full 1.0/100% feels more worthwhile IMO.

gfwilliams, 2025-02-24:
For comparison:

```
var a = new Uint8Array(256);
for (var i=0;i<4096;i++) a[0|(255*i/4095)]++;
a = new Uint8Array([17, 16, 16, 16, 16,  ... 16, 16, 16, 16, 1]);
// or if you times by 256 you can one value that's out of range

var a = new Uint8Array(256);
for (var i=0;i<4096;i++) a[0|(256*i/4096)]++;
a = new Uint8Array([16, 16, 16, 16, 16,  ... 16, 16, 16, 16, 16]);
```

commit 87286c56 (2025-02-24): STM32: Ensure proper distribution of analogRead values (now `0<=analogRead()<1` as on other boards) and add docs (fix #2612)
closed 2025-02-24 by gfwilliams