espruino/Espruino #2613: Problem with SPI on Pico

mariusGundersen, 2025-03-03:
I've found a weird issue with the SPI on the Pico (might be others too, but that's where I'm testing it). My setup is a pic together with the [Adafruit eInk Feather Friend](https://www.adafruit.com/product/4446), so that I can read images from an SD card and show them on an E-ink display. This means using the SPI to read from the SD card using one chip-select pin and then writing to the e-ink display using another chip-select. I have tested and gotten both the reading from SD card and updating the e-ink display to work, but not together. The error I get is this:

```
Uncaught InternalError: Timeout on SPI RX
```

I've searched the code for the string "SPI RX" and found only one potential place it can happen: https://github.com/espruino/Espruino/blob/e6b2fd24bd2feef447299097be8d6e65e09f4a7d/targets/stm32/jshardware.c#L2470

This happens when I try to read from the SD card after writing to the e-ink display. I'm using the chip-select to toggle which device I want to communicate with, but something still seems to go wrong. After a lot of trial and error and experimenting I found a workaround: changing from `spi.write(data, cs)` to `spi.send(data, cs)` when communicating with the e-ink display. The display has no MISO pin, so I've been using `write()` since there is no data coming back. But it seems like this is messing with the internal SPI state machine. The code for recreating this is fairly simple:

```js

SPI1.setup({
    sck: B3,
    miso: B4,
    mosi: B5,
});

E.connectSDCard(SPI1, A5);
require("fs").readdirSync(''); // this works, returns a list of files

// write to SPI1 but using another chip select pin
SPI1.write(0x10, A6)

require("fs").readdirSync(''); // now it fails

// setting up the SPI and SD card again fixes the problem
SPI1.setup({
    sck: B3,
    miso: B4,
    mosi: B5,
});

E.connectSDCard(SPI1, A5);
require("fs").readdirSync(''); // this works, returns a list of files
```

It seems like the write code is messing up the read code, but I'm unable to debug any deeper than this.

mariusGundersen, 2025-03-03:
I'm fairly sure I'm on the latest, yes. I haven't tried another version

gfwilliams, 2025-03-03:
I've been able to reproduce without actually setting up an SD card on the Pico, but I'm pretty sure I found the issue.

I've put a fix in, so I think if you try a cutting edge build now it should be fixed?

mariusGundersen, 2025-03-03:
yes, works great with the latest build, thanks!

commit efeeb30e (2025-03-03): STM32: Fix spi.write usage on same port that SD card is used with (fix #2613)