espruino/Espruino #2614: PIPBOY_LINUX build appears broken since adding PIN_TEARING

rikkuness, 2025-03-16:
Hey, I may be incorrect in my findings but I've been unable to build against board `PIPBOY_LINUX` as it fails with:
```
libs/pipboy/jswrap_pipboy.c:1134:18: error: ‘LCD_TEARING’ undeclared (first use in this function)
 1134 |   jshPinSetState(LCD_TEARING, JSHPINSTATE_GPIO_IN_PULLDOWN);
```

This reference seems to have been introduced in https://github.com/espruino/Espruino/commit/01aa8f0d5e5dc2f287cf7b0be52b31d6bc14f444

From my digging this definition is added by a `pin_tearing` param of the `devices.LCD` in the board definition, which is present in the `PIPBOY` board definition and adding a dummy value into the `PIPBOY_LINUX.py` does seem to make that error go away. 

Alterntively the cleaner option may be to add a conditional here?
```diff
index f2957b67c..b058cb67d 100644
--- a/libs/pipboy/jswrap_pipboy.c
+++ b/libs/pipboy/jswrap_pipboy.c
@@ -1132,7 +1132,9 @@ static void jswrap_pb_periph_off() {
   jshPinSetState(DAC_SDA_PIN, JSHPINSTATE_GPIO_IN);
   jshPinSetState(RADIO_SCL_PIN, JSHPINSTATE_ADC_IN);
   jshPinSetState(RADIO_SDA_PIN, JSHPINSTATE_ADC_IN);
+#ifdef LCD_TEARING
   jshPinSetState(LCD_TEARING, JSHPINSTATE_GPIO_IN_PULLDOWN);
+#endif
   STM32_I2S_Kill();
 }
```

Assume @rblakesley should be tagged for review 😄 

Thanks <3

closed 2025-03-17 by gfwilliams

gfwilliams, 2025-03-17:
Ahh, ok - the PIPBOY_LINUX build was something we did for debugging pretty early on (when for instance we wanted to be able to test/iterate with video decoding very quickly) - it's not something that may be that much use in itself I'm afraid - for instance it doesn't support any audio output, or any of the button inputs.

It might actually be a target it makes sense to remove I'm afraid

rikkuness, 2025-03-17:
Yeha honestly I didn't really have any specific use case for it, I was just curious and digging around 😄 I did get it built with some modifications and got to the "NO SD CARD" screen and had the Espruino interactive console etc. but outside those of use that just like to hack on things yeah I suppose there's not a ton of use 😆

commit 80fb12b4 (2025-03-17): Fix PIPBOY_LINUX build - fix #2614