espruino/Espruino #2615: Assert fails at line 835 of jsvar.c on ESP32-S3

GaryOtt, 2025-03-20:
This snipped from a [benchmarking code](https://github.com/orgs/espruino/discussions/7710):

```
let rf = (d) => {
	var myObj = {};
	for (let a = 0; a < 80; a += 1) myObj["param" + a] = a;

	if (d < 5) { rf(d + 1); rf(d + 1); }
};

rf(0);
```

causes:

```

 ____                 _ 
|  __|___ ___ ___ _ _|_|___ ___ 
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v25 (c) 2024 G.Williams

Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate

>let rf = (d) => {
:var myObj = {};
:for (let a = 0; a < 80; a += 1) myObj["param" + a] = a;
:
:if (d < 5) { rf(d + 1); rf(d + 1); }
:};
=function (d) { ... }

assert failed: jsvLockAgain jsvar.c:835 (jsvGetLocks(var) < JSV_LOCK_MAX)

Backtrace: 0x40375fde:0x3fcbaf50 0x40383639:0x3fcbaf70 0x4038ba8d:0x3fcbaf90 0x420116fa:0x3fcbb3

ELF file SHA256: c2addc3f3517e136

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375ca8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x16c8
load:0x403c9700,len:0xbd4
load:0x403cc700,len:0x2fb0
entry 0x403c9954

```

This does not happen on the ESP32 or any of the official boards.

gfwilliams, 2025-03-20:
It's because you're not compiling with RELEASE=1 - this happens on any board

I think there might be an issue for this, but we count the number of locks - if you do something weird like referencing the same object 80 times then the lock counter overflows and wraps over. In reality it's not that big a deal - as long as the lock counter is nonzero you're fine - and it's very unlikely that you'll reference something just enough times to make the lock count overflow and be zero.

It's not ideal, but it's worked pretty well so far. The alternative would be keeping the lock counter static at the maximum allowed value if it hit it, which would avoid issues but would introduce a variable that then couldn't ever be freed.

closed 2025-03-20 by gfwilliams

gfwilliams, 2025-03-20:
Just an update - that's not 100% correct on my part as that would be references and this is Locks (but I think it's the recursion that gets you in this case).

Either way in retrospect it's probably worth sacrificing some speed/memory to ensure that we can't get something weird like this happening so I created an issue and I'll try and get some kind of fix in.

GaryOtt, 2025-03-20:
I usually always build with either RELEASE=1 or DEBUG=1. I accept I must have omitted both on this occasion. I'll build again and see what happens.

The time at which this assert fails is interesting. It gets to the 5th level of recursion, so has created 4 objects each with 80 properties already. And when it attempts to add the first property to the 5th object, it fails. If too many properties on an object is the root cause of the problem, then I would have thought it would fail sooner.

gfwilliams, 2025-03-20:
The locks occur when the interpreter has a pointer to the given variable - so you're right, in this case the 80 references aren't the problem (there was a previous similar issue with references but looking at it that got fixed a while back).

The issue is the recursion I believe - I just looked into it and it's actually the Flat String that holds the function code that's the issue - and the max number of locks is 15. I'll have a look tomorrow as it might be jspeFunctionCall is holding onto a `functionCode` lock longer than it really needs to and we could at least get 50% more recursion (maybe more) by just reordering a few things.

GaryOtt, 2025-03-20:
I have rebuilt Espruino for the S3 using the commands below. Still I am getting an assert failure. Have I done something wrong or is there a new issue here that RELEASE=1 isn't removing assert statements?

```
cd Espruino
source scripts/provision.sh ESP32S3_IDF4
make clean
make RELEASE=1 BOARD=ESP32S3_IDF4
```

gfwilliams, 2025-03-21:
I think the order is wrong - you need then before `make` -  try: `RELEASE=1 BOARD=ESP32S3_IDF4 make`