espruino/Espruino #2617: ESP32-C3 non-responsive console over USB serial

GaryOtt, 2025-03-20:
This afternoon, I tried to play with an ESP32-C3 for the first time. I took a new and previously unused ESP32-C3-WROOM-O2 from its box and flashed Espruino onto it.

I tried to use a serial connection but found that while I could get serial output from the board, nothing I typed was echoed or responded to.

I was able to get responses using the Web Bluetooth capability of https://www.espruino.com/ide/

Having not used the board before, I wrote a simple Arduino sketch to echo data received by serial connection. The board is fully working.

SimonGAndrews, 2025-03-21:
Garry , Hi,  the ESP32C3 port has an option to build Espruino for enabling the console on either the Serial RXD/TXD Pins or the USB (JTAG Driver) D+/D-  pins.    The default build in the Espruino distribution is for the USB Pins.

[image]

If you are using the Serial UART you can build a version after a small modification to the board definition file (ESP32C3_IDF4.)  The mod is to comment out the definition of  DUSB_CDC as seen in this pic (not commented out).

[image]

Else you can utilise the USB Pins with the distributed version.

This method is currently subject to a change thats being worked by myself to fix issue #2609 .

BTW Are you using the Serial Pins?   Are you using the raw module or is it on a board (Which one) ?  Are you building your own versions of Espruino?
Hope tis helps.   Simon

GaryOtt, 2025-03-21:
Thank you for your help, @SimonGAndrews 

I will try this.

gfwilliams, 2025-03-21:
So you have a link to the board you used?

As @SimonGAndrews says it's set up to work over the on-chip USB by default, but some boards don't connect that and instead have a separate USB-TTL chip - in which case right now those won't work out the box.

gfwilliams, 2025-03-21:
Thanks! Yes - I can see the CP2102 chip on there, so that's your problem.

Hopefully the work @SimonGAndrews is doing will sort this out though

GaryOtt, 2025-03-21:
Yes, that worked. Thank you both. I'll close this as duplicate.

closed 2025-03-21 by GaryOtt

SimonGAndrews, 2025-03-21:
Great news.  
I  have an Espressif ESP32-C3-Devkitc-02  board and will include in the testing of #2609
Cheers