espruino/Espruino #2618: ESP32-C3 fails to kick watchdog during long running tasks

GaryOtt, 2025-03-20:
After failing to use the [ESP32-C3 over serial](https://github.com/espruino/Espruino/issues/2617), I tried to use a Web Bluetooth connection instead.

After starting a [long running task](https://github.com/orgs/espruino/discussions/7710), I did not get the response I expected. I tried again while observing the serial connection and found that the board reset because it had not kicked the watchdog.

gfwilliams, 2025-03-21:
Ok, thanks. I thought we'd disabled the task watchdog, but it seems not: https://github.com/espruino/Espruino/blob/master/targets/esp32/IDF4/sdkconfig_c3#L815-L818

I know it doesn't work well for your benchmarks (maybe using `E.kickWatchdog()` in them might help?) but there's an argument that normally code wouldn't hang for 5sec, and maybe keeping the watchdog isn't such a bad idea?

SimonGAndrews, 2026-01-20:
Suggest this is closed as a repeat of #2588 .  Ive just updated that discussion to reflect the current status of the watchdog timer on ESP32C3 build.

closed 2026-01-21 by gfwilliams