espruino/Espruino #2637: Calculation error on BangleJS1 with Firmware 2v26

nxdefiant, 2025-05-29:
Just updated my BangleJS1 to Firmware 2v26 and got this calculation anomaly on Math.sin() with negative numbers:

> >process.env
={
  VERSION: "2v26",
  GIT_COMMIT: "c83c27c02",
  BOARD: "BANGLEJS",
  RAM: 65536, FLASH: 524288, SPIFLASH: 4194304, HWVERSION: 1, STORAGE: 4194304,
  SERIAL: "d767714f-7aece0e9",
  CONSOLE: "Bluetooth",
  MODULES: "Flash,Storage,hea" ... "tensorflow,locale",
  EXPTR: 480356 }
>Math.sin(-6.283)
=-25.37477787616
>Math.sin(6.283)
=-0.00018530717

nxdefiant, 2025-06-02:
Unfortunatly there is still an error:
`Math.sin(-240*Math.PI/180)`
> =-0.86602538208

gfwilliams, 2025-06-03:
Sorry, I'm an idiot. Math.sin isn't symmetrical around 0 is it?

I've fixed it now, and actually tested it works ok too ;)