espruino/Espruino #2641: Bangle.js buzz when back widget tapped

RKBoss6, 2025-07-10:
For me, the native back widget seemed to be quite finicky to use, and I had no idea why. Having a bit of haptic feedback (`Bangle.buzz()`) when the back widget is clicked would be a really lovely feature to give haptic feedback that you touched the button, hopefully preventing frustrated touches in the area until you get to it. I looked over the SetWatch function, and the back widget seems to be implemented in there, but as this is firmware related stuff, I've got no idea on how to test that part without needing to rebuild the firmware version each time. Does anyone know how to do this safely?
Thanks!

I also feel as though this might be part of a larger update, adding more haptic feedback, for instance, when you long press and release, a buzz to show that it's going to the clock would be good.

gfwilliams, 2025-07-11:
That sounds like it could be a good idea. You can change the built-in functions by overriding them in JS - there's a tutorial on it at https://www.espruino.com/Bangle.js+Modification which might be a good place to start and test things? Then when it's ok we can push the changes to firmware

I think you need to modify setUI - there is already some code in there for haptic feedback, it just doesn't appear to be rolled out to everything: https://github.com/espruino/Espruino/blob/master/libs/js/banglejs/Bangle_setUI_Q3.js#L38-L40

Handling the long-press is a bit trickier - that's possibly something that'd have to be in the firmware.

gfwilliams, 2025-07-11:
I'm not sure what you mean about the back function, but the touch handler is at https://github.com/espruino/Espruino/blob/master/libs/js/banglejs/Bangle_setUI_Q3.js#L132-L134

Literally just add `b()` to that function

RKBoss6, 2025-07-11:
@gfwilliams Alright, I attempted to create a PR for review.

RKBoss6, 2025-07-11:
I think an app could achieve this with more flexibility though, what do you think?