espruino/Espruino #2649: ESP32-S3 - Unknown Pins 40-48

TheoCGaming, 2025-07-23:
I'm attempting to drive the built-in LCD on the Waveshare ESP32-S3-Touch-LCD-2.8 development board, but the source code that the Waveshare examples use uses pins beyond 39 to drive the LCD. Part of the issue is that I don't know what pin labels 40-48 are supposed to correspond to.

# Excerpt of C Header code from `ESP32-S3-Touch-LCD-2.8-Demo/ESP-IDF/ESP32-S3-Touch-LCD-2.8-Test/main/LCD_Driver/ST7789.h`:

```
#include "freertos/task.h"
#include "esp_timer.h"
#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_vendor.h"
#include "esp_lcd_panel_ops.h"
#include "driver/gpio.h"
#include "driver/spi_master.h"
#include "esp_err.h"
#include "esp_log.h"
#include "lvgl.h"
#include "driver/ledc.h"

#include "Vernon_ST7789T.h"
#include "CST328.h"
#include "LVGL_Driver.h"
// LCD SPI GPIO
// Using SPI2 
#define LCD_HOST  SPI3_HOST

#define EXAMPLE_LCD_PIXEL_CLOCK_HZ     (80 * 1000 * 1000)
#define EXAMPLE_LCD_BK_LIGHT_ON_LEVEL  1
#define EXAMPLE_LCD_BK_LIGHT_OFF_LEVEL !EXAMPLE_LCD_BK_LIGHT_ON_LEVEL
#define EXAMPLE_PIN_NUM_MISO           -1
#define EXAMPLE_PIN_NUM_MOSI           45
#define EXAMPLE_PIN_NUM_SCLK           40
#define EXAMPLE_PIN_NUM_LCD_CS         42
#define EXAMPLE_PIN_NUM_LCD_DC         41
#define EXAMPLE_PIN_NUM_LCD_RST        39
#define EXAMPLE_PIN_NUM_BK_LIGHT       5
// The pixel number in horizontal and vertical
#define EXAMPLE_LCD_H_RES              240
#define EXAMPLE_LCD_V_RES              320
// Bit number used to represent command and parameter
#define EXAMPLE_LCD_CMD_BITS           8
#define EXAMPLE_LCD_PARAM_BITS         8

#define Offset_X 0
#define Offset_Y 0
```

# Modified example JS code from ST7789 library:

```
const LCD_WIDTH = 240;
const LCD_HEIGHT = 320;

// ST7789 demo code on TTGO T-Display
D4.set(); // LCD backlight on
var spi = new SPI();
spi.setup({mosi:45, sck:40});
var g = require("ST7789").connect(spi, 41, 42, 39, function() {
    g.clear();
    g.setRotation(1);
    g.drawString("Hello",0,0);
    g.setFontVector(20);
    g.setColor(0,0.5,1);
    g.drawString("Espruino",0,10);
  });
```

# Espruino console output:

```
ERROR: pinToESP32Pin: Unknown pin: 40
ERROR: pinToESP32Pin: Unknown pin: 41
ERROR: pinToESP32Pin: Unknown pin: 41
ERROR: pinToESP32Pin: Unknown pin: 42
ERROR: pinToESP32Pin: Unknown pin: 42
ERROR: pinToESP32Pin: Unknown pin: 43
ERROR: pinToESP32Pin: Unknown pin: 43
ERROR: pinToESP32Pin: Unknown pin: 44
ERROR: pinToESP32Pin: Unknown pin: 44
ERROR: pinToESP32Pin: Unknown pin: 45
ERROR: pinToESP32Pin: Unknown pin: 46
ERROR: pinToESP32Pin: Unknown pin: 46
ERROR: pinToESP32Pin: Unknown pin: 47
ERROR: pinToESP32Pin: Unknown pin: 47
ERROR: pinToESP32Pin: Unknown pin: 48
ERROR: pinToESP32Pin: Unknown pin: 48
WARNING: Not removing services for reset() as connected
WARNING: Skip jsble_set_services as connected
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v27 (c) 2025 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
>
ERROR: pinToESP32Pin: Unknown pin: 40
ERROR: pinToESP32Pin: Unknown pin: 45
Uncaught Error: Function "reset" not found!
    at g (:1:3)
c.reset();a.write(b,f);void 0!==e&&(c.set(),a.write(e,f))
  ^
    at :1:5
g(17);setTimeout(function(){d.forEach(function(b){g(b[0],b[1...
    ^
>
```

SimonGAndrews, 2025-07-23:
It looks like the function gpio_num_t pinToESP32Pin in esp32/jshardwate.c is hardcoded checking the pin count for the original ESP32 device only and not the newer ESP32 devices.  It is currently 
``` c#
/**
 * Convert an Espruino pin id to a native ESP32 pin id.
 */
gpio_num_t pinToESP32Pin(Pin pin) {
  if ( pin < 40 )
    return pin + GPIO_NUM_0;
  jsError( "pinToESP32Pin: Unknown pin: %d", pin);
  return -1;
}
```

Something like this may  be better:

``` c#
extern const uint8_t available_pins[];
extern const size_t num_available_pins;

/**
 * Convert an Espruino pin id to a native ESP32 pin id.
 * Enforces runtime validation against board definition.
 */
gpio_num_t pinToESP32Pin(Pin pin) {
  for (size_t i = 0; i < num_available_pins; ++i) {
    if (available_pins[i] == pin) {
    return pin + GPIO_NUM_0;
    }
  }
  jsError("pinToESP32Pin: Invalid or undefined pin: %d", pin);
  return -1;
}
```
I dont currently have a clean S3 build to test this.  But will try ASAP

SimonGAndrews, 2025-07-23:
PS - However Im happy to test  on ESP32, C3 and S3,  if the change is cut in.

SimonGAndrews, 2025-07-23:
Sorry the above suggestion does not build.   Looking at an alternative suggestion

SimonGAndrews, 2025-07-23:
so 
``` c#
#include "jspin.h" // for jshIsPinValid()

gpio_num_t pinToESP32Pin(Pin pin) {
  if (jshIsPinValid(pin)) {
    return (gpio_num_t)pin;
  }

  jsError("pinToESP32Pin: Invalid or undefined pin: %d", pin);
  return -1;
}
```

gfwilliams, 2025-07-24:
Thanks for looking into this - `jshIsPinValid` should be the one. If that returns true for a `Pin`, it is valid,

Is there more than GPIO_NUM_0 for ESP32S3?

If so, in `pinInfo` we have `JSH_PORT_*` which are supposed to be for each GPIO port.

Also usually you'd also do:

```
gpio_num_t pinToESP32Pin(Pin pin) {
  if (jshIsPinValid(pin)) {
    return pinInfo[pin].pin + GPIO_NUM_0;
  }

  jsError("pinToESP32Pin: Invalid or undefined pin: %d", pin);
  return -1;
}
```

It allows people to create builds where the actual pin number on the board doesn't match the CPU's pin number (eg the board manufacturer maybe chose to rename pins)

TheoCGaming, 2025-07-25:
> `return pinInfo[pin].pin + GPIO_NUM_0;`

does this mean that I can address raw pin numbers through (theoretically) GPIO_NUM_0 through GPIO_NUM_48, excluding some in the 20s?

SimonGAndrews, 2025-07-31:
@TheoCGaming , Hi, Exactly Which S3 board are you using ?    Ill try to test with the same if I have one.   Because there are still some issues with a generic s3 build around the presence/use of a uart or not.

SimonGAndrews, 2025-07-31:
Good news,  thanks, I’ll test on the other boards before sending a PR to G.

MaBecker, 2026-09-15:
This now merged and fixed

closed 2026-09-15 by MaBecker