espruino/Espruino #2651: looking for efficient analog read and processing

ZacharyLightman, 2025-08-02:
I am trying to use a MDBT42Q to modify an analog signal. I built a my own board that features a MCP4727 DACto provide analog output, which works very well. I have been trying to take the analog input and apply a rather complicated transfer function and then send it back out using the mcp4728 dac.  my transfer function includes some exponents and because of this the code execution time is about 12ms from input change to output change. I was thinking that instead of using the analog value , a fraction between 0 to 1 , i should instead convert the fractional value to an integer and then i can run the math much faster, especially if i use the native JIT compiler. The input and output should both be 12bit, so this is totally reasonable for a 16 bit word to handle.  

I'm just asking for is the most efficient way to convert the analog reading to a integer with espruino. My goal is the get the execution time below 10 milliseconds. /i would appreciate any help on this matter as i am always interested into how to best utilize the espruino hardware and firmware.  
I am not sure if using WAVEFORMS would be the best route, but it did feel like overkill to sample at Khz rates. I

gfwilliams, 2025-08-04:
Ok, so you want the time from input to output to be as low as possible, rather than specifically worrying about sample rate?

`Waveform` is good if you want a steady sample rate, but then it's not so good at passing that data through. Internally the value from the ADC comes in as an integer and stays as an integer, getting written to the array though.

JIT could help, but right now the JIT compiler calls through to general purpose maths functions even if the numbers are known to be integers, so while you will see some speed improvements from using ints, it's not going to be huge.

Honestly, if you really want to get the processing time down, I'd look at using Inline C (https://www.espruino.com/InlineC). You could access the ADC directly in C code (using the NRF52LL library as an example: https://github.com/espruino/EspruinoDocs/blob/master/devices/NRF52LL.js#L216-L273) and then you can be sure you keep everything in integers, in compiled code.

(btw I think questions like this are probably best asked in the forum? https://github.com/orgs/espruino/discussions)

ZacharyLightman, 2025-08-04:
Thank you for the suggestions. I will probably try to re-write portions of my code so i can speed up portions of the code, and perhaps compile it. I used Jit to hit 10ms execution time without forcing everything to an integer so i'm pretty happy with that. 
And yea, i should put my discussions in the discussions area, huh, oops.

gfwilliams, 2025-08-05:
Great! While floating point is slower, there is a lot of overhead from the JS execution, so as you note the JIT can really help. If you're willing to share your code I might be able to provide some suggestions. Some things (like accessing fields) can still be slower, so for example you can often look them up just once at the start and get a decent speed boost:

```
// slow
for (var i=0;i<100;i++) ...Main.sin(x)
// faster
var sin = Math.sin;
for (var i=0;i<100;i++) ...sin(x)
```

ZacharyLightman, 2025-08-05:
this code allows me to change the transfer function during operation for a control system. that is why I'm trying to use global variables. It is nice to be able to interrogate the variables arbitrarily.  I also output the input and output so I can monitor(and display) them on a graph.   Using JIT the execution time is down to an average of 9.5ms now, with about 21 ms without JIT.  This code is still a work in progress but this is what i have:

`//Global Settings
var data = [0,0,0];
var voltage = [0,0];
var inOffset = 0.0;
var inSpan = 0.6;//abritrary Waveform generator only goes to 2v peak to peak 
var outOffset = 0.0;
var outSpan = 1;
//var i =  I2C1;
const MCP4728_ADDR = 0x60;
//Global Math Variables 
//for initiate seemless variable change (update variables) 
var change = false;
var enable = true;
// exponent value of p-norm function(smoothed transfer function) 
var p = 7;
//slope of initial slope function 
var gM = 0.25;
var TgM = 0.25;
//intercept for initial slope funtion 
var gB = 0.18;
var TgB = 0.18;
// slope of 2nd slope funtion 
var fM = 0.89;
var TfM = 0.89;
//intercept of 2nd slope function 
var fB = -0.09;
var TfB = -0.09;
//Hysteresis Quotient
var Q = 2;
var TQ = 2;
//Intercept for Hysteresis Function 
var I = 0.3;
var TI = 0.3;
//Y is the stored output value
var Y = 0.18;
// Globals for output strings
var out_now = 0;
var in_now = 0;
//var s ="";
//configure I2C for communication with ADC chip
I2C1.setup({ scl : D16, sda: D17 }); ////for this example
function onInit(){
  //On startup change the BLE Advertising name 
	NRF.setAdvertising({},{name: "DrivePad SW 5.0"});
  setup();}
function setup(){
//start running using intervals
setInterval(loop, 5); // aspirational goal for execution time!
//start reporting values
setInterval(send, 20); //report values every 20ms to the console, adds ~1ms for execution
}

function checkVars(){
  if (TgM != gM){ gM = TgM;}

  if (TfM != fM){
    fM = TfM;
    fB = FindfB(fM);
    }
  if (TQ != Q){ Q = TQ;}
  if (TI!= I){ I = TI;}

}

function restart(){
  clearInterval();
  onInit();
}

function maths(I){
  "jit";
  I=I*4095;
  if (I < 0){I=0;}
  out = I | 0;
  return out;
}
function setMCP4728Channels(valA, valB) {
  
  "jit";
  // Ensure values are 12-bit
  //valA = Math.max(0, Math.min(4095, valA));
  //valB = Math.max(0, Math.min(4095, valB));
  // Build the Write DAC and EEPROM command
  // Multi-write command: C2 C1 C0 = 0b010 -> Write DAC Input Register
  // Command format: 0x50 | (C2..C0)
  const cmd = 0x40;  // Write DAC input registers sequentially starting at channel A

  // Pack values: each channel is 3 bytes
  const data = [
    0x41, // Write DAC input registers sequentially starting at channel A 0x40+1 for UDAC
    ((valA >> 8) & 0x0F), // upper nibble contains 4 MSBs of val
    valA & 0xFF,                       // lower 8 bits
      0x43,// Channel B command 0x42 +1 for UDAC
    ((valB >> 8) & 0x0F), // config bits + upper nibble
    valB & 0xFF
  ];

  // Write to device
  I2C1.writeTo(MCP4728_ADDR, data);
}

//output scaling for ADC
function output(o){
if (o <= 0.18){ o = 0.18;}
else if (o >=0.80){o = 0.80;}
else { o = o;}
o = o * outSpan + outOffset;
out_now = o; //send to global variable for transmission later
if (!enable){ o = 0.18 * outSpan + outOffset;}//change the output variable back to the minimum after the value is sent to phone app. 
out = maths(o);
setMCP4728Channels(out, out);
}

function hysteresis(x){
     "jit";
     tY = Y;
     px = P(x,fM,fB,gM,gB,p);
     pi = P(I,fM,fB,gM,gB,p);
     hx = H(px,pi,Q);
  if (x>I){
    if ( Y < px){tY = px;}
        else if (Y > hx ){tY = hx;}
    }
  else{tY = px;}
  return tY;
}
//start reading input 
function input(inOff,inSp){
"jit";
inn = (analogRead(D28)-inOff)/inSp;
if (inn <= 0){inn = 0;}
else if (inn >= 1){inn = 1;}
else { inn = inn;}
return inn;

}
function loop(){
  "jit";
  if (change) {checkVars();}
inn = input(inOffset,inSpan);
in_now = inn;
Y = hysteresis(inn);
output(Y);
}

//function G(x){  return (gM*x + gB);}

//function F(x){   return (fM*x + fB);}

function FindfB(m){return (0.8 - m );}

function H(PX,PI,q){
"jit";
return PX*q+(PI-(PI*q));
}

//function P(x) { return Math.pow(Math.pow(F(x),p)+Math.pow(G(x),p),1/p);} 
function P(x,fm,fb,gm,gb,pwr) { 
  //function P(x) { return Math.pow(Math.pow(F(x),p)+Math.pow(G(x),p),1/p);} 
  "jit";
  return Math.pow(Math.pow((fm*x + fb),pwr)+Math.pow((gm*x + gb),pwr),1/pwr);
} 
//(Math.pow(Math.pow((fm*x + fb),pwr)+Math.pow((gm*x + gb),pwr),1/pwr))
function send(){
s= "[0,"+in_now.toFixed(3)+","+out_now.toFixed(3)+"]";
console.log(s);
}`

this is testing utilizing a PICOSCOPE testing the latency between input and output , the blue Triangle wave is the input and the red in the output. : 

[image]

gfwilliams, 2025-08-05:
Thanks for posting the code!

I can see a few easy things that might help:

* in `input` you use `inn` which is a global but you return it anyway. If you use `var inn =` it'll be a local and should go faster
* The less function calls you have the better, so if you can put `input` and `hysteresis` in `loop` that'll help
* you're using a lot of globals, like fM/fB/etc. If you put those in an object like `state.fM/etc` and then pass that object into the function, that should be faster as it's only having to look them up within the object